commaai/openpilot #23985: Openpilot in Carla isn't accelerating: (engaged: True ; throttle:  0.0)

ebadi, 2022-03-17:
### Describe the bug

There is known issue with modeld crashing because of libzmq library with [the known solution](https://github.com/commaai/openpilot/issues/23666) that **used to work a few weeks ago**.  It seems that recent changes introduced a new issue. Now openpilot, while engaged, is not accelerating in the CARLA simulator (engaged: True ; throttle:  0.0)

shell 1:
[image]
shell 2:
[image]

### OS Version

Ubuntu 18.04

### openpilot version or commit

db7b49c71cfc420259d99d3a4a6f0972b4f887e8

ebadi, 2022-03-17:
It is possible to use AWSD keys to control the car in CARLA from `bridge.py`. So it seem the issue is with openpilot module.

ebadi, 2022-03-23:
I tried an old **working** version of the repository and here are the _important_ changes that I see when I run `./launch_openpilot.sh`:

```
selfdrive/locationd/locationd.cc: Locationd vs ubloxLocation position difference too large, kalman reset
...
{"event": "commIssue", "invalid": [], "not_alive": ["deviceState", "pandaStates", "peripheralState", "modelV2", "liveCalibration", "longitudinalPlan", "lateralPlan", "liveLocationKalman", "liveParameters", "radarState", "roadCameraState", "driverCameraState"], "can_error": false, "error": true}
...
selfdrive/modeld/modeld.cc: connected with buffer size: 3493536 (1928 x 1208)
...
ready to run onnx model ['input_imgs', 'desire', 'traffic_convention', 'initial_state'] [[1, 12, 128, 256], [1, 8], [1, 2], [1, 512]]
{"event": "commIssue", "invalid": [], "not_alive": ["driverCameraState"], "can_error": true, "error": true}
```

Ming2888, 2022-03-24:
@ebadi , where did you find the old working version? I have the same trouble running Openpilot in Carla.

ebadi, 2022-03-24:
@Ming2888 If you only want to get it running, clone this repo and follow the instruction: https://github.com/jackhong12/openpilot

Other relevant posts:
https://github.com/commaai/openpilot/issues/23695
https://github.com/commaai/openpilot/pull/23792

jackhong12, 2022-03-24:
The scalars of the timestamp are different in real-world and simulation, which causes an infinite loop in modeld.

In the `selfdrive/modeld/modeld.cc`, it keeps receiving data until the timestamp of the meta_main is 25000000 more than the timestamp of the meta_extra. However, the timestamp value is much smaller than 25000000 in simulation. So the loop below will cause modeld to hang. By the way, does 25000000 mean 0.025 second?
 ```c++
    while (get_ts(meta_main) < get_ts(meta_extra) + 25000000ULL) {
      buf_main = vipc_client_main.recv(&meta_main);
      if (buf_main == nullptr)  break;
    }
```

The timestamp is assigned in `tools/sim/bridge.py`.  I think one way to fix this issue is multiplied `eof` by a constant value.
```cpp
    # convert RGB frame to YUV
    rgb = np.reshape(img, (H, W * 3))
    rgb_cl = cl_array.to_device(self.queue, rgb)
    yuv_cl = cl_array.empty_like(rgb_cl)
    self.krnl(self.queue, (np.int32(self.Wdiv4), np.int32(self.Hdiv4)), None, rgb_cl.data, yuv_cl.data).wait()
    yuv = np.resize(yuv_cl.get(), np.int32(rgb.size / 2))
    eof = self.frame_id * 0.05

    # TODO: remove RGB send once the last RGB vipc subscriber is removed
    self.vipc_server.send(VisionStreamType.VISION_STREAM_RGB_BACK, img.tobytes(), self.frame_id, eof, eof)
    self.vipc_server.send(VisionStreamType.VISION_STREAM_ROAD, yuv.data.tobytes(), self.frame_id, eof, eof)

    dat = messaging.new_message('roadCameraState')
    dat.roadCameraState = {
      "frameId": image.frame,
      "transform": [1.0, 0.0, 0.0,
                    0.0, 1.0, 0.0,
                    0.0, 0.0, 1.0]
    }
    pm.send('roadCameraState', dat)
    self.frame_id += 1
```

After modifying the value of `eof`, Openpilot can accelerate again.
- https://github.com/jackhong12/openpilot/commit/d014d8191d950a56a961c18c84463ffd0d3d5de5

ebadi, 2022-03-24:
Thanks @jackhong12. I tried your patch on the master branch and I still keep getting this error (with or without your fix)
```Onnx available providers:  ['TensorrtExecutionProvider', 'CUDAExecutionProvider', 'CPUExecutionProvider']
Onnx selected provider:  ['CUDAExecutionProvider']
2022-03-24 21:08:17.173280866 [W:onnxruntime:Default, onnxruntime_pybind_state.cc:535 CreateExecutionProviderInstance] Failed to create CUDAExecutionProvider. Please reference https://onnxruntime.ai/docs/reference/execution-providers/CUDA-ExecutionProvider.html#requirements to ensure all dependencies are met.
Onnx using  ['CPUExecutionProvider']
ready to run onnx model ['input_imgs', 'big_input_imgs', 'desire', 'traffic_convention', 'initial_state'] [[1, 12, 128, 256], [1, 12, 128, 256], [1, 8], [1, 2], [1, 512]]
Traceback (most recent call last):
  File "/openpilot/selfdrive/modeld/runners/onnx_runner.py", line 60, in <module>
    run_loop(ort_session)
  File "/openpilot/selfdrive/modeld/runners/onnx_runner.py", line 34, in run_loop
    inputs.append(read(ts).reshape(shp))
  File "/openpilot/selfdrive/modeld/runners/onnx_runner.py", line 17, in read
    assert(len(st) > 0)
AssertionError
clocksd logcatd
```

maykonpacheco, 2022-03-25:
@ebadi have you tried doing this in your branch?

[image]

Ming2888, 2022-03-25:
@maykonpacheco, your suggestion works for me. Thank you!
Somehow my car still still drive off the road after about one minute , as mentioned in issue https://github.com/commaai/openpilot/issues/22265.

ebadi, 2022-03-25:
Thanks @jackhong12, I tested it again and now it is working. I upvoted your pull request https://github.com/commaai/openpilot/pull/24034 and I hope get merged to the master soon. Once it get merged I will write few tests to avoid the simulation environment to break again.
 

@maykonpacheco I tried the fix on the master branch and for some reason I got those error messages. I though the issue might be that I picked a wrong branch as you see in this page: https://github.com/commaai/openpilot/commit/d014d8191d950a56a961c18c84463ffd0d3d5de5. Anyway it is now working :). 

[image]

closed 2022-03-25 by ebadi