commaai/openpilot #24016: Openpilot in Simulation Mode is not using IMU data.

rahul-arasikere, 2022-03-21:
### Describe the bug

When running the bridge.py script to bridge the Carla server, in the IMU callback, the sensorEvent packet does not set a timestamp, and hence in locationd, the packet is rejected as it checks the timestamp to 0, and rejects the packet as it assumes there is no valid data.

I am not sure if in simulation mode this is the expected behaviour or a bug and wanted to point it out.

A possible solution would be to extend common.clock.pyx to include a `nanosec_since_boot()` function as the carla server maintains simulation time apart from linux.

### OS Version

Ubuntu 20.04

tiredDev, 2022-04-04:
Can I start working on an issue or does it need to be assigned first ?

pd0wm, 2022-04-20:
Feel free to open a draft PR once you made some progress. We don't assign issues.

equilibrium-liberalism, 2022-08-18:
> ### Describe the bug
>
> When running the bridge.py script to bridge the Carla server, in the IMU callback, the sensorEvent packet does not set a timestamp, and hence in locationd, the packet is [...]

Have you encountered following issue when running bridge.py ? sorry for a entry level question. 

**case 1'**
~/openpilot/tools/sim$ python3 bridge.py
Traceback (most recent call last):
  File "bridge.py", line 19, in <module>
    import cereal.messaging as messaging
ModuleNotFoundError: No module named 'cereal'

**case 2**
:~/openpilot/tools/sim$     ./bridge.py
Traceback (most recent call last):
  File "./bridge.py", line 11, in <module>
    import carla  # pylint: disable=import-error
ModuleNotFoundError: No module named 'carla'

rahul-arasikere, 2022-08-19:
You need to run the python script from the main `openpilot` folder, or update `$PYTHONPATH` to include the openpilot directory. `carla` is another python package you need along with the carla server.

equilibrium-liberalism, 2022-08-20:
@rahul-arasikere Thank you for your reply. it enlightened me as a new bee of openpilot.

adeebshihadeh, 2022-09-30:
fixed in #25937

closed 2022-09-30 by adeebshihadeh