commaai/openpilot #24022: CAN Error and LKAS Fault when in Dashcam Only Mode

Verylukyguy, 2022-03-22:
### Describe the bug

I loaded the current OpenPilot Release2 and on startup in Dashcam OnIy Mode I immediately got a CAN Error and LKAS Fault.

### What hardware does this issue affect?

comma two

### Provide a route where the issue occurs

e38a02cae1ef1e0c|2022-03-22--18-18-30--0

### openpilot version

v0.8.13-release

adeebshihadeh, 2022-03-23:
Is your ASCM unplugged?

Verylukyguy, 2022-03-23:
> Is your ASCM unplugged?
Yes, the ASCM is bypassed.

adeebshihadeh, 2022-03-23:
You'll have to plug any unplugged ECUs back in for dashcam mode to work. Dashcam mode is completely passive and doesn't send anything on the CAN bus.

Verylukyguy, 2022-03-23:
Okay, Sorry, I wasn't sure if this was a new issue.

Thanks.

Try not to forget about my ongoing Steering Errors and LKAS Faults, I'd love to use stock OpenPilot eventually.

closed 2022-03-23 by adeebshihadeh