commaai/openpilot #24029: BUILD FAILED (OS X 12.3 using python-build 20180424)

royjr, 2022-03-24:
### Describe the bug

Tried building on clean M1 Mac using `tools/mac_setup.sh` and `ARCHFLAGS="-arch arm64" ./tools/mac_setup.sh`

### OS Version

macOS Monterey 12.3 (21E230)

### Additional info

```
Homebrew Bundle complete! 23 Brewfile dependencies now installed.
[ ] finished brew install t=1
python 3.8.10 install ...
python-build: use openssl@1.1 from homebrew
python-build: use readline from homebrew
Downloading Python-3.8.10.tar.xz...
-> https://www.python.org/ftp/python/3.8.10/Python-3.8.10.tar.xz
Installing Python-3.8.10...
python-build: use readline from homebrew
python-build: use zlib from xcode sdk

BUILD FAILED (OS X 12.3 using python-build 20180424)

Inspect or clean up the working tree at /var/folders/64/36h4jv0j2ss6dl91pgvqdgv40000gn/T/python-build.20220323223505.73551
Results logged to /var/folders/64/36h4jv0j2ss6dl91pgvqdgv40000gn/T/python-build.20220323223505.73551.log

Last 10 log lines:
checking for --with-cxx-main=<compiler>... no
checking for clang++... no
configure:

  By default, distutils will build C++ extension modules with "clang++".
  If this is not intended, then set CXX on the configure command line.
  
checking for the platform triplet based on compiler characteristics... darwin
configure: error: internal configure error for the platform triplet, please file a bug report
make: *** No targets specified and no makefile found.  Stop.
```

```
cat /var/folders/64/36h4jv0j2ss6dl91pgvqdgv40000gn/T/python-build.20220323223505.73551.log
/var/folders/64/36h4jv0j2ss6dl91pgvqdgv40000gn/T/python-build.20220323223505.73551 ~/Documents/Github/openpilot
/var/folders/64/36h4jv0j2ss6dl91pgvqdgv40000gn/T/python-build.20220323223505.73551/Python-3.8.10 /var/folders/64/36h4jv0j2ss6dl91pgvqdgv40000gn/T/python-build.20220323223505.73551 ~/Documents/Github/openpilot
checking build system type... arm-apple-darwin21.4.0
checking host system type... arm-apple-darwin21.4.0
checking for python3.8... no
checking for python3... python3
checking for --enable-universalsdk... no
checking for --with-universal-archs... no
checking MACHDEP... "darwin"
checking for gcc... clang
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether clang accepts -g... yes
checking for clang option to accept ISO C89... none needed
checking how to run the C preprocessor... clang -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for a sed that does not truncate output... /usr/bin/sed
checking for --with-cxx-main=<compiler>... no
checking for clang++... no
configure:

  By default, distutils will build C++ extension modules with "clang++".
  If this is not intended, then set CXX on the configure command line.
  
checking for the platform triplet based on compiler characteristics... darwin
configure: error: internal configure error for the platform triplet, please file a bug report
make: *** No targets specified and no makefile found.  Stop.
```

royjr, 2022-03-24:
Changed .python-version from 3.8.10 to 3.9.10. Build somewhat successful, but scons -u -j$(nproc) fails.

<details><summary>tools/mac_setup.sh logs</summary>
<p>

```
Using catch2
Using cmake
Using cppcheck
Using git-lfs
Using zlib
Using bzip2
Using capnp
Using coreutils
Using eigen
Using ffmpeg
Using glfw
Using libarchive
Using libusb
Using libtool
Using llvm
Using openssl
Using pyenv
Using qt@5
Using zeromq
Using protobuf
Using protobuf-c
Using swig
Using gcc-arm-embedded
Homebrew Bundle complete! 23 Brewfile dependencies now installed.
[ ] finished brew install t=2
python 3.9.10 install ...
python-build: use openssl@1.1 from homebrew
python-build: use readline from homebrew
Downloading Python-3.9.10.tar.xz...
-> https://www.python.org/ftp/python/3.9.10/Python-3.9.10.tar.xz
Installing Python-3.9.10...
python-build: use readline from homebrew
python-build: use zlib from xcode sdk
Installed Python-3.9.10 to /Users/admin/.pyenv/versions/3.9.10

update pip
Collecting pip==21.3.1
  Downloading pip-21.3.1-py3-none-any.whl (1.7 MB)
     |████████████████████████████████| 1.7 MB 3.4 MB/s 
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 21.2.4
    Uninstalling pip-21.2.4:
      Successfully uninstalled pip-21.2.4
Successfully installed pip-21.3.1
Collecting pipenv==2021.11.23
  Downloading pipenv-2021.11.23-py2.py3-none-any.whl (3.6 MB)
     |████████████████████████████████| 3.6 MB 3.4 MB/s            
Collecting certifi
  Downloading certifi-2021.10.8-py2.py3-none-any.whl (149 kB)
     |████████████████████████████████| 149 kB 30.3 MB/s            
Collecting virtualenv
  Downloading virtualenv-20.13.4-py2.py3-none-any.whl (8.7 MB)
     |████████████████████████████████| 8.7 MB 25.1 MB/s            
Requirement already satisfied: setuptools>=36.2.1 in /Users/admin/.pyenv/versions/3.9.10/lib/python3.9/site-packages (from pipenv==2021.11.23) (58.1.0)
Collecting virtualenv-clone>=0.2.5
  Downloading virtualenv_clone-0.5.7-py3-none-any.whl (6.6 kB)
Requirement already satisfied: pip>=18.0 in /Users/admin/.pyenv/versions/3.9.10/lib/python3.9/site-packages (from pipenv==2021.11.23) (21.3.1)
Collecting distlib<1,>=0.3.1
  Downloading distlib-0.3.4-py2.py3-none-any.whl (461 kB)
     |████████████████████████████████| 461 kB 23.9 MB/s            
Collecting filelock<4,>=3.2
  Downloading filelock-3.6.0-py3-none-any.whl (10.0 kB)
Collecting platformdirs<3,>=2
[... 2787 lines cut ...]
panda/board/drivers/timers.h: In function 'interrupt_timer_init':
panda/board/drivers/timers.h:21:3: internal compiler error: Illegal instruction: 4
   21 |   register_set(&(INTERRUPT_TIMER->PSC), ((uint16_t)(15.25*APB1_FREQ)-1U), 0xFFFFU);
      |   ^~~~~~~~~~~~
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.linaro.org/> for instructions.
scons: *** [panda/board/sha-panda.o] Error 1
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.linaro.org/> for instructions.
scons: *** [panda/board/main-panda_h7.o] Error 1
during GIMPLE pass: cddce
panda/crypto/rsa.c: In function 'geM':
panda/crypto/rsa.c:294:1: internal compiler error: Illegal instruction: 4
  294 | }
      | ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.linaro.org/> for instructions.
scons: *** [panda/board/rsa-panda_h7.o] Error 1
scons: building terminated because of errors.
```
</p>
</details>

fpinnola, 2022-03-24:
I am having a similar issue building on M1 Macbook. Setup step works, `scons -u -j$(nproc)` fails. OS X version 12.0.1. Likely an architecture issue. 

```
In file included from panda/board/stm32fx/stm32fx_config.h:62,
                 from panda/board/config.h:34,
                 from panda/board/bootstub.c:7:
panda/board/drivers/timers.h: In function 'interrupt_timer_init':
panda/board/drivers/timers.h:21:3: internal compiler error: Illegal instruction: 4
   21 |   register_set(&(INTERRUPT_TIMER->PSC), ((uint16_t)(15.25*APB1_FREQ)-1U), 0xFFFFU);
      |   ^~~~~~~~~~~~
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.linaro.org/> for instructions.
scons: *** [panda/board/bootstub-panda.o] Error 1
In file included from panda/board/stm32h7/stm32h7_config.h:54,
                 from panda/board/config.h:32,
                 from panda/board/bootstub.c:7:
panda/board/drivers/timers.h: In function 'interrupt_timer_init':
panda/board/drivers/timers.h:21:3: internal compiler error: Illegal instruction: 4
   21 |   register_set(&(INTERRUPT_TIMER->PSC), ((uint16_t)(15.25*APB1_FREQ)-1U), 0xFFFFU);
      |   ^~~~~~~~~~~~
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.linaro.org/> for instructions.
scons: *** [panda/board/bootstub-panda_h7.o] Error 1
during GIMPLE pass: cddce
panda/crypto/rsa.c: In function 'geM':
panda/crypto/rsa.c:294:1: internal compiler error: Illegal instruction: 4
  294 | }
      | ^
In file included from panda/board/stm32fx/stm32fx_config.h:62,
                 from panda/board/config.h:34,
                 from panda/board/main.c:2:
panda/board/drivers/timers.h: In function 'interrupt_timer_init':
panda/board/drivers/timers.h:21:3: internal compiler error: Illegal instruction: 4
   21 |   register_set(&(INTERRUPT_TIMER->PSC), ((uint16_t)(15.25*APB1_FREQ)-1U), 0xFFFFU);
      |   ^~~~~~~~~~~~
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.linaro.org/> for instructions.
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.linaro.org/> for instructions.
during GIMPLE pass: evrp
panda/crypto/sha.c: In function 'SHA1_Transform':
panda/crypto/sha.c:179:1: internal compiler error: Illegal instruction: 4
  179 | }
      | ^
scons: *** [panda/board/rsa-panda.o] Error 1
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.linaro.org/> for instructions.
scons: *** [panda/board/main-panda.o] Error 1
scons: *** [panda/board/sha-panda.o] Error 1
during GIMPLE pass: cddce
panda/crypto/rsa.c: In function 'geM':
panda/crypto/rsa.c:294:1: internal compiler error: Illegal instruction: 4
  294 | }
      | ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.linaro.org/> for instructions.
scons: *** [panda/board/rsa-panda_h7.o] Error 1
In file included from panda/board/stm32h7/stm32h7_config.h:54,
                 from panda/board/config.h:32,
                 from panda/board/main.c:2:
panda/board/drivers/timers.h: In function 'interrupt_timer_init':
panda/board/drivers/timers.h:21:3: internal compiler error: Illegal instruction: 4
   21 |   register_set(&(INTERRUPT_TIMER->PSC), ((uint16_t)(15.25*APB1_FREQ)-1U), 0xFFFFU);
      |   ^~~~~~~~~~~~
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.linaro.org/> for instructions.
during GIMPLE pass: evrp
panda/crypto/sha.c: In function 'SHA1_Transform':
panda/crypto/sha.c:179:1: internal compiler error: Illegal instruction: 4
  179 | }
      | ^
scons: *** [panda/board/main-panda_h7.o] Error 1
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.linaro.org/> for instructions.
scons: *** [panda/board/sha-panda_h7.o] Error 1
scons: building terminated because of errors.
```

lskov, 2022-03-29:
I experienced the same issue recently after update to latest xcode command line tool 13.3
You can downgrade to 13.2 and the it will work.

First make sure to remove /Library/Developer/CommandLineTools
Then download version 13.2 of Xcode build tools and install this
Finally make sure to run xcode-select -s <your path to new installed version>

royjr, 2022-03-30:
Thanks @lskov, that fixed `tools/mac_setup.sh`, but now I'm having the same issue as @fpinnola 

<details><summary>scons -u -j$(nproc) logs</summary>
<p>

```
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
clang++ -o opendbc/can/packer_pyx.so -bundle -undefined dynamic_lookup /Users/admin/Documents/Github/openpilot/opendbc/can/libdbc.dylib opendbc/can/packer_pyx.o -Lthird_party/libyuv/mac_arm64/lib -L/opt/homebrew/lib -L/opt/homebrew/Library -L/opt/homebrew/opt/openssl/lib -L/opt/homebrew/Cellar -Lthird_party/acados/Darwin/lib -L/System/Library/Frameworks/OpenGL.framework/Libraries -Lcereal -Lthird_party -Lopendbc/can -Lselfdrive/boardd -Lselfdrive/common
clang++ -o opendbc/can/parser_pyx.so -bundle -undefined dynamic_lookup /Users/admin/Documents/Github/openpilot/opendbc/can/libdbc.dylib opendbc/can/parser_pyx.o -Lthird_party/libyuv/mac_arm64/lib -L/opt/homebrew/lib -L/opt/homebrew/Library -L/opt/homebrew/opt/openssl/lib -L/opt/homebrew/Cellar -Lthird_party/acados/Darwin/lib -L/System/Library/Frameworks/OpenGL.framework/Libraries -Lcereal -Lthird_party -Lopendbc/can -Lselfdrive/boardd -Lselfdrive/common
arm-none-eabi-gcc -o panda/board/bootstub-panda.o -c -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -T/Users/admin/Documents/Github/openpilot/panda/board/stm32fx/stm32fx_flash.ld -std=gnu11 -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Os -g -DPANDA -DALLOW_DEBUG -Ipanda/board/stm32fx/inc -Ipanda/board/stm32h7/inc -Ipanda -Ipanda/board panda/board/bootstub.c
arm-none-eabi-gcc -o panda/board/bootstub-panda_h7.o -c -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -T/Users/admin/Documents/Github/openpilot/panda/board/stm32h7/stm32h7x5_flash.ld -std=gnu11 -mcpu=cortex-m7 -mhard-float -DSTM32H7 -DSTM32H725xx -mfpu=fpv5-d16 -fsingle-precision-constant -Os -g -DPANDA -DALLOW_DEBUG -Ipanda/board/stm32fx/inc -Ipanda/board/stm32h7/inc -Ipanda -Ipanda/board panda/board/bootstub.c
arm-none-eabi-gcc -o panda/board/main-panda.o -c -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -T/Users/admin/Documents/Github/openpilot/panda/board/stm32fx/stm32fx_flash.ld -std=gnu11 -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Os -g -DPANDA -DALLOW_DEBUG -Ipanda/board/stm32fx/inc -Ipanda/board/stm32h7/inc -Ipanda -Ipanda/board panda/board/main.c
arm-none-eabi-gcc -o panda/board/main-panda_h7.o -c -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -T/Users/admin/Documents/Github/openpilot/panda/board/stm32h7/stm32h7x5_flash.ld -std=gnu11 -mcpu=cortex-m7 -mhard-float -DSTM32H7 -DSTM32H725xx -mfpu=fpv5-d16 -fsingle-precision-constant -Os -g -DPANDA -DALLOW_DEBUG -Ipanda/board/stm32fx/inc -Ipanda/board/stm32h7/inc -Ipanda -Ipanda/board panda/board/main.c
arm-none-eabi-gcc -o panda/board/rsa-panda.o -c -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -T/Users/admin/Documents/Github/openpilot/panda/board/stm32fx/stm32fx_flash.ld -std=gnu11 -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Os -g -DPANDA -DALLOW_DEBUG -Ipanda/board/stm32fx/inc -Ipanda/board/stm32h7/inc -Ipanda -Ipanda/board panda/crypto/rsa.c
arm-none-eabi-gcc -o panda/board/sha-panda.o -c -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -T/Users/admin/Documents/Github/openpilot/panda/board/stm32fx/stm32fx_flash.ld -std=gnu11 -mcpu=cortex-m4 -mhard-float -DSTM32F4 -DSTM32F413xx -mfpu=fpv4-sp-d16 -fsingle-precision-constant -Os -g -DPANDA -DALLOW_DEBUG -Ipanda/board/stm32fx/inc -Ipanda/board/stm32h7/inc -Ipanda -Ipanda/board panda/crypto/sha.c
arm-none-eabi-gcc -o panda/board/rsa-panda_h7.o -c -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -T/Users/admin/Documents/Github/openpilot/panda/board/stm32h7/stm32h7x5_flash.ld -std=gnu11 -mcpu=cortex-m7 -mhard-float -DSTM32H7 -DSTM32H725xx -mfpu=fpv5-d16 -fsingle-precision-constant -Os -g -DPANDA -DALLOW_DEBUG -Ipanda/board/stm32fx/inc -Ipanda/board/stm32h7/inc -Ipanda -Ipanda/board panda/crypto/rsa.c
arm-none-eabi-gcc -o panda/board/sha-panda_h7.o -c -Wall -Wextra -Wstrict-prototypes -Werror -mlittle-endian -mthumb -nostdlib -fno-builtin -T/Users/admin/Documents/Github/openpilot/panda/board/stm32h7/stm32h7x5_flash.ld -std=gnu11 -mcpu=cortex-m7 -mhard-float -DSTM32H7 -DSTM32H725xx -mfpu=fpv5-d16 -fsingle-precision-constant -Os -g -DPANDA -DALLOW_DEBUG -Ipanda/board/stm32fx/inc -Ipanda/board/stm32h7/inc -Ipanda -Ipanda/board panda/crypto/sha.c
In file included from panda/board/stm32fx/stm32fx_config.h:62,
                 from panda/board/config.h:34,
                 from panda/board/bootstub.c:7:
panda/board/drivers/timers.h: In function 'interrupt_timer_init':
panda/board/drivers/timers.h:21:3: internal compiler error: Illegal instruction: 4
   21 |   register_set(&(INTERRUPT_TIMER->PSC), ((uint16_t)(15.25*APB1_FREQ)-1U), 0xFFFFU);
      |   ^~~~~~~~~~~~
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.linaro.org/> for instructions.
In file included from panda/board/stm32h7/stm32h7_config.h:54,
                 from panda/board/config.h:32,
                 from panda/board/bootstub.c:7:
[... cut ...]
scons: *** [panda/board/rsa-panda_h7.o] Error 1
scons: *** [panda/board/main-panda_h7.o] Error 1
during GIMPLE pass: evrp
panda/crypto/sha.c: In function 'SHA1_Transform':
panda/crypto/sha.c:179:1: internal compiler error: Illegal instruction: 4
  179 | }
      | ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.linaro.org/> for instructions.
scons: *** [panda/board/sha-panda_h7.o] Error 1
scons: building terminated because of errors.
```

</p>
</details>

zorrobyte, 2022-05-06:
Alternatively, Python could be updated to .13 (from .10) which resolves this issue

khoi, 2022-05-12:
temporary fix for the panda compiling problem

```
brew uninstall gcc-arm-embedded
curl -L https://github.com/Homebrew/homebrew-cask/blob/d407663b8017a0a062c7fc0b929faf2e16abd1ff/Casks/gcc-arm-embedded.rb > 1.rb
brew install 1.rb
```

zorrobyte, 2022-05-12:
This should fix it https://github.com/commaai/openpilot/pull/24515

Thanks @khoi

closed 2022-05-13 by pd0wm