commaai/openpilot #24031: Having the C3 plugged in (running openpilot) breaks remote start

cawilliamson, 2022-03-24:
Hello!

So I just had remote start installed on my vehicle (in the UK - we don't get it from factory like the US folks!) via a Pandora alarm system. Essentially it emulates a human - holds the break, hits the start button, yada yada.

Now it's installed - it works 100% of the time when I don't have the C3 (running OP) plugged in but as soon as I plug it in - it starts to fail the process. I believe the C3 is querying the car the second it gets power and slowing down things on the car just enough to cause the remote start to fail (a race condition.)

Obviously this is an annoying issue so what I'm thinking is - would it be possible to either:

1. Add an option in stock OP to delay comms with the vehicle for a period of time?
2. Fork OP and add this option myself?
3. Add some kind of bash script type workaround somewhere in OP (since the C3 runs Ubuntu anyway) to intentionally slow this down?

Any help here would be greatly appreciated as unplugging / re-plugging the C3 every time is getting old real fast! :)

Thanks!

pd0wm, 2022-03-28:
Do you have a drive where the remote start fails?

Sounds like the alarm system also tries to do some UDS queries, but doesn't implement proper retry logic. Since this issue is very rare I doubt we want to fix this on the openpilot side. 

As a workaround you can disable fingerprinting using environment variables `FINGERPRINT` and `SKIP_FW_QUERY`: https://github.com/commaai/openpilot/blob/master/selfdrive/car/car_helpers.py#L92. However, this might break things like DSU detection so YMMV.

adeebshihadeh, 2022-04-12:
Closing due to inactivity.

closed 2022-04-12 by adeebshihadeh