commaai/openpilot #24040: modeld cannot recover when bridge is terminated

ebadi, 2022-03-25:
### Describe the bug

When bridge.py is terminated, it seems that modeld should keep trying until a frame is received. (https://github.com/commaai/openpilot/blob/master/selfdrive/modeld/modeld.cc#L89)
However modeld stucks in printing `vipc_client_main no frame` error messages even after bridge.py is restarted. 

### OS Version

Ubuntu 18.04

HaraldSchafer, 2022-06-02:
I could not reproduce this. When bridge is killed, openpilot goes offroad so it kills modeld anyway? When bridge restarted modeld starts from scratch?

HaraldSchafer, 2022-06-02:
Closing for now

closed 2022-06-02 by HaraldSchafer

adeebshihadeh, 2022-06-02:
Probably fixed as a side effect of 302b0ea9bbac0fec939919b0d09b604569305be7