commaai/openpilot #24049: Compile issues seen on Ubuntu 18

crsumiran, 2022-03-27:
### Describe the bug

I see 11 compile errors, some pointing to "byte" type data and some pointing to certain macros. I am trying to compile the code and run it on my laptop. Can you please help me at the earliest. I tried changing std to 17 and also tried the solution from a suggested solution for the "byte" problem (https://github.com/catchorg/Catch2/issues/1992). I am unable to find the cause and I am not sure about the other errors because they seem to be repository specific.

In file included from selfdrive/ui/replay/filereader.cc:6:
./selfdrive/ui/replay/util.h:34:38: error: no type named 'byte' in namespace 'std'
std::string decompressBZ2(const std::byte *in, size_t in_size, std::atomic<bool> *abort = nullptr);
                                ~~~~~^
1 error generated.
scons: *** [selfdrive/camerad/camera-filereader.o] Error 1
In file included from selfdrive/ui/replay/framereader.cc:1:
./selfdrive/ui/replay/framereader.h:23:24: error: no type named 'byte' in namespace 'std'
  bool load(const std::byte *data, size_t size, bool no_hw_decoder = false, std::atomic<bool> *abort = nullptr);
                  ~~~~~^
./selfdrive/ui/replay/framereader.h:34:28: error: unknown type name 'AVHWDeviceType'
  bool initHardwareDecoder(AVHWDeviceType hw_device_type);
                           ^
In file included from selfdrive/ui/replay/framereader.cc:2:
./selfdrive/ui/replay/util.h:34:38: error: no type named 'byte' in namespace 'std'
std::string decompressBZ2(const std::byte *in, size_t in_size, std::atomic<bool> *abort = nullptr);
                                ~~~~~^
selfdrive/ui/replay/framereader.cc:81:15: error: reference to type 'const std::string' (aka 'const basic_string<char>') could not bind to an rvalue of type 'std::byte *'
  return load((std::byte *)data.data(), data.size(), no_hw_decoder, abort);
              ^~~~~~~~~~~~~~~~~~~~~~~~
selfdrive/ui/replay/framereader.cc:76:43: note: passing argument to parameter 'url' here
bool FrameReader::load(const std::string &url, bool no_hw_decoder, std::atomic<bool> *abort, bool local_cache, int chunk_size, int retries) {
                                          ^
selfdrive/ui/replay/framereader.cc:84:19: error: out-of-line definition of 'load' does not match any declaration in 'FrameReader'
bool FrameReader::load(const std::byte *data, size_t size, bool no_hw_decoder, std::atomic<bool> *abort) {
                  ^~~~
selfdrive/ui/replay/framereader.cc:126:30: error: use of undeclared identifier 'AV_HWDEVICE_TYPE_CUDA'
    if (!initHardwareDecoder(HW_DEVICE_TYPE)) {
                             ^
selfdrive/ui/replay/framereader.cc:13:24: note: expanded from macro 'HW_DEVICE_TYPE'
#define HW_DEVICE_TYPE AV_HWDEVICE_TYPE_CUDA
                       ^
selfdrive/ui/replay/framereader.cc:153:39: error: unknown type name 'AVHWDeviceType'
bool FrameReader::initHardwareDecoder(AVHWDeviceType hw_device_type) {
                                      ^
selfdrive/ui/replay/framereader.cc:155:11: error: unknown type name 'AVCodecHWConfig'
    const AVCodecHWConfig *config = avcodec_get_hw_config(decoder_ctx->codec, i);
          ^
selfdrive/ui/replay/framereader.cc:155:37: error: use of undeclared identifier 'avcodec_get_hw_config'
    const AVCodecHWConfig *config = avcodec_get_hw_config(decoder_ctx->codec, i);
                                    ^
selfdrive/ui/replay/framereader.cc:161:27: error: use of undeclared identifier 'AV_CODEC_HW_CONFIG_METHOD_HW_DEVICE_CTX'
    if (config->methods & AV_CODEC_HW_CONFIG_METHOD_HW_DEVICE_CTX && config->device_type == hw_device_type) {
                          ^
selfdrive/ui/replay/framereader.cc:227:16: error: use of undeclared identifier 'av_hwframe_transfer_data'
    if ((ret = av_hwframe_transfer_data(hw_frame.get(), av_frame_.get(), 0)) < 0) {
               ^
11 errors generated.

### OS Version

Ubuntu 18.04

GregorKikelj, 2022-03-27:
Openpilot isn’t supported on Ubuntu 18, it's only expected to work on 20.04.

adeebshihadeh, 2022-03-28:
@GregorKikelj is right. We target the Ubuntu LTS used on our workstations and AGNOS, currently 20.04, and will merge fixes for the very latest Ubuntu LTS. Beyond that, we make no guarantees.

closed 2022-03-28 by adeebshihadeh