commaai/openpilot #24069: Civic Nidec: noisy standstill signal

sshane, 2022-03-29:
The standstill signal can be very noisy at times for no reason:

[image]

spektor56, 2022-03-30:
> The standstill signal can be very noisy at times for no reason, so we end up cancelling sometimes if you take your foot off the brake at a stop:
> 
> [image]

Looks like you are looking at standstill and not cruisestate.standstill?

sshane, 2022-03-30:
CruiseState describes the car's cruise control state, we want the car's state.

spektor56, 2022-03-30:
> CruiseState describes the car's cruise control state, we want the car's state. It's what we use for controls

It's not specific to nidec though, my Bosch CRV has the same fuzzyness in OPs standstill variable.  The cruisestate one is just used for the resume spam when using stock ACC

sshane, 2022-07-07:
Fixed in https://github.com/commaai/openpilot/pull/24262