commaai/openpilot #24083: honda: acceleration oscillations after reaching target speed

ajRiverav, 2022-03-30:
### Describe the bug

Bug can be reproduced/observed on mostly flat roads. 
The humans in the car feel minor acceleration oscillations which coincide with the tachometer going slightly up and down.
One interesting observation is that when it does happen, comma 3 shows an actual speed of 1 mph lower than car's (e.g. car shows 64 but openpilot shows 65). 

It feels like a classic PID controller oscillation.

### What hardware does this issue affect?

comma three

### Which car does this affect?

Honda Odyssey 2019

### Provide a route where the issue occurs

0c23034a3a1e7bf4|2022-03-29--12-23-57--0

### openpilot version

0.8.13

adeebshihadeh, 2023-07-06:
Still having this issue?

ajRiverav, 2023-07-23:
no

closed 2023-07-23 by ajRiverav