commaai/openpilot #24087: Add hidden toggle to disable internet connectivity check

JasonAHeron, 2022-03-31:
TLDR: Can we add a custom toggle to disable the internet DAYS_NO_CONNECTIVITY_MAX

Hey folks,
I mentioned this today in the comma clubhouse. I have a custom fork that I maintain because my wife does not have weekly connection to internet with her car. https://github.com/JasonAHeron/openpilot/commit/f8b67dbacac67559a9a20fea2625a9882a143267

However I still want to contribute driving data so Geohot mentioned that a custom toggle would be a way to solve this problem without me needing to create a custom toggle.

gregjhogan, 2022-03-31:
setting `DisableUpdates` hidden param doesn't handle this?

adeebshihadeh, 2022-03-31:
That's also an option if you want to completely disable the updater.

JasonAHeron, 2022-03-31:
Seems like ```rm /data/params/d/LastUpdateTime``` might be the path for me.

My use case is that I want to continue to get updates and be able to contribute driving data but not have any internet connectivity requirements due to the fact that my wife can't reliably park anywhere with internet and has no tethering on her phone.