commaai/openpilot #24110: Unfortunately, Android Keyboard (AOSP) has stopped

aquamammal, 2022-04-02:
### Describe the bug

I reflashed the Comma 2. It was successful, however, now I can't use the android: This pops up "Unfortunately, Android Keyboard (AOSP) has stopped."

No matter how many times I reflash from the github. 

### What hardware does this issue affect?

comma two

### Provide a route where the issue occurs

I am reflashing the Comma 2, following the instructions on the github, and have it reflashed, and upon turning on, it says "Unfortunately, Android Keyboard (AOSP) has stopped."

### openpilot version

0.8.10

abdulrahman004, 2022-04-03:
have you tried restarting device ?

adeebshihadeh, 2022-04-05:
Have you tried reflashing the device with https://github.com/commaai/eon-neos?

adeebshihadeh, 2022-04-06:
Closing this since it's not an openpilot bug. If you still have the issue after reflashing NEOS, check out the community discord for help.

closed 2022-04-06 by adeebshihadeh