commaai/openpilot #24112: UI for TOS and Training keep appearing

ebadi, 2022-04-04:
### Describe the bug

UI for TOS and Training keep appearing when the docker is restarted.
I tried to fix the issue by [modifying manager.py](https://github.com/commaai/openpilot/blob/master/selfdrive/manager/manager.py#L39) with no success.
```
  if PC :
    default_params: List[Tuple[str, Union[str, bytes]]] = [
    ("CompletedTrainingVersion", "1"),
    ("HasAcceptedTerms", "1"),
    ("OpenpilotEnabledToggle", "1"),
    ]
  else:
    default_params: List[Tuple[str, Union[str, bytes]]] = [
    ("CompletedTrainingVersion", "0"),
    ("HasAcceptedTerms", "0"),
    ("OpenpilotEnabledToggle", "1"),
    ]
```

### OS Version

Ubuntu 18.04

### openpilot version or commit

eaf66b7e083c821ecbf9613c834dca9e24ccf1de

adeebshihadeh, 2022-04-04:
Should be fixed by  #24113. If you'd like to test, you'll need to rebuild the docker image or wait for it to automatically be built and pushed after merging into master.

adeebshihadeh, 2022-04-04:
Merged the fix into master and the image should be updated. Let me know if this fixed it for you.

ebadi, 2022-04-04:
I tested this and it works. Thanks @adeebshihadeh.
p.s. Are you interested if I write a simple unit-test to ensure openpilot can drive inside Carla? If so, any preference what has to be tested?

closed 2022-04-04 by ebadi

adeebshihadeh, 2022-04-04:
Definitely interested in tests around the simulator. See #2572. There's also this open PR #24009.

ebadi, 2022-04-06:
I took a look at it today @adeebshihadeh.  I wonder why the open PR has not been merged? Would it make sense to wait until this PR is merged first?