commaai/openpilot #24121: Set missing car minimum enable/steer speeds in CarParams

sshane, 2022-04-04:
Some cars don't have any minimum enable or steer speeds defined in their CarParams as we expect the car to handle this for us if we're `pcmCruise`. So we can eventually remove these exceptions: https://github.com/commaai/openpilot/blob/e40c36f22ba93200a5fc83211005b28fca3fa728/selfdrive/car/docs_definitions.py#L55

Draft PRs to work off of:

- https://github.com/commaai/openpilot/pull/23942
- https://github.com/commaai/openpilot/pull/23941

abesisu, 2022-10-03:
High there @sshane, I'm a student working on a class at Iowa State University where we choose an open-source project to analyze and contribute to. If I were to try and contribute to this issue, how might I find the correct missing minimum enable/steer speeds that need to be put in?

abesisu, 2022-10-18:
@sshane is there anything I'm doing wrong in [this PR](https://github.com/commaai/openpilot/pull/26136) that would lead these checks to fail? I haven't changed CarInfo and I've only added to CarParams. I originally tried to remove the min_steer_speed from values.py, but the test I'm failing said the minSteerSpeed should be in both CarParams and CarInfo.

jyoung8607, 2022-10-18:
> @sshane is there anything I'm doing wrong in [this PR](https://github.com/commaai/openpilot/pull/26136) that would lead these checks to fail? I haven't changed CarInfo and I've only added to [...]

The unit test is complaining _because_ both are set; they shouldn't be. The CarInfo setting should only be used if there's some reason they aren't (or can't) be set in CarParams; now that you have set them, they should be removed from CarInfo.

The process replay test is mainly a speed bump designed to make sure openpilot behaves the same before and after any given change, and if it doesn't, that someone examines the difference and confirms that it's expected and desired. In your case, it detects the CarParams change. Assuming the change has no ill effects in real world testing on those cars, a comma employee would review that difference and update the replay baseline to include them as part of merging the PR.

abesisu, 2022-10-18:
@jyoung8607 do you know of a way that I can get the unit test to pass? If what I've done is correct? https://github.com/commaai/openpilot/pull/26136

jyoung8607, 2022-10-18:
For the docs unit tests, when you removed the min steer speed from the individual HondaCarInfo instances, the default min steer speed set in the [top level HondaCarInfo class](https://github.com/commaai/openpilot/blob/826d8a8ae34bfd909535f5edff2229b8b2daf1a6/selfdrive/car/honda/values.py#L105-L108) replaced it. That HondaCarInfo default will need to go away if you're going to drive these settings via carParams, taking care to implement carParams for all Hondas or setting min speed explicitly at the individual HondaCarInfo instances.

For the process replay tests, for your purposes, don't worry about it failing for now.

jimbrend, 2023-12-20:
why has this gone stale?

adeebshihadeh, 2024-02-20:
Still relevant @sshane?

closed 2024-04-15 by adeebshihadeh