commaai/openpilot #24122: HKG: Verify and remove cars from torque blacklist

sshane, 2022-04-04:
Need to verify if these cars can send up to 384 instead of 255 for the max torque:

- [ ] Genesis G80 2018
- [ ] Genesis G90 2018
- [ ] Hyundai Elantra 2017-19
- [ ] Hyundai Genesis 2015-16
- [ ] Hyundai Ioniq Hybrid 2017-19
- [ ] Hyundai Ioniq Electric 2019
- [ ] Hyundai Santa Fe Plug-in Hybrid 2022
- [ ] Hyundai Sonata 2018-19
- [ ] Kia Forte 2018-21
- [ ] Kia Niro Plug-in Hybrid 2019
- [ ] Kia Optima Hybrid
- [ ] Kia Sorento 2018-19
- [ ] Kia Stinger 2018

adeebshihadeh, 2022-04-04:
@sunnyhaibin any ones you already verified that we can bring over from your fork?

sunnyhaibin, 2022-04-05:
> @sunnyhaibin any ones you already verified that we can bring over from your fork?

@adeebshihadeh Yes. Below is the list of cars that were confirmed to be able to send up to 384 for the max torque on my fork:
- Hyundai Elantra 2017-19
- Hyundai Ioniq Hybrid 2017-19
- Hyundai Ioniq Electric 2019
- Hyundai Santa Fe Plug-in Hybrid 2022
- Kia Niro Plug-in Hybrid 2019

Some of the cars on the issue's car list are using my fork as well. I will get them to test and report back.

adeebshihadeh, 2022-04-05:
Great, that's a good chunk of the remaining ones!

sunnyhaibin, 2022-04-07:
@adeebshihadeh The following car is also confirmed to work well with sending up to 384 for the max torque:
- Genesis 2015 and 2016

sshane, 2022-04-07:
Do you have any routes where these cars are commanding that torque? I can throw a branch together with them on the higher torque

sunnyhaibin, 2022-04-09:
> Do you have any routes where these cars are commanding that torque? I can throw a branch together with them on the higher torque

That would be great if you can setup a branch for them.

I believe some users have their devices banned due to previous violations, but I'll see what I can do.

sshane, 2022-04-12:
Let us know if you have anyone in the above list try this branch: https://github.com/commaai/openpilot/commits/h384

sshane, 2022-04-13:
It's possible we need to reduce the steering tune when we merge these cars, @Jon8RFC on Discord noticed jerking for turns with the new higher steering torque (no faults).

sunnyhaibin, 2022-04-22:
@sshane Here's a route with the `Hyundai Ioniq Electric 2019`!
- `8e221cf40bc9c78a|2022-04-20--17-51-52`

multikyd, 2022-05-20:
Those cars should be okay with 384 torque including my car, Optima Hybrid 2017(MY16 K5 Hybrid from Korea). I've never heard there has been issues related to 384 from HKG old&new cars. Every HKG users will be happy with the value. Thank you Comma!

sshane, 2022-05-20:
We will probably wait until the torque controller is added back, as raising the max torque likely means the car needs a new tune for optimal control

gregjhogan, 2022-07-18:
I recall people saying some cars accept values higher than 255, but not getting additional torque above 255, so if we increase it we should verify we are not just adding a dead-zone at the top end of the request value range.  This should be possible by looking at the motor output torque vs request value.

I found this related conversation:
https://discord.com/channels/469524606043160576/524611978208215070/737805997992706059
```
At that time any value above 255, up to 470 something, resulted in the same torque output.
```

adeebshihadeh, 2023-02-01:
Community has been doing a pretty good job of testing and moving cars over - don't think we need an issue for this.

closed 2023-02-01 by adeebshihadeh

j2mc, 2024-11-14:
I see this was closed, but it appears most of the cars in this list are still being limited.  Has there been more testing done to verify these cars need to be limited?  I have a 22 Santa Fe PHEV and would be happy to provide any data that's needed.