commaai/openpilot #24124: Subaru Fingerprinting Issues

michaelhonan, 2022-04-05:
### Describe the bug

 I've noticed that my car that previously FP'd as `SUBARU IMPREZA 2017` is now being recognised as mock or SUBARU FORESTER 2019.

Here's a dump of my carFw: They're in slightly different orders, but all addresses are accounted for and the fw versions match between all three

[image]

SUBARU IMPREZA 2017 Route (Actual car): dc30ef2ad294a732|2022-04-04--13-04-36
SUBARU FORESTER 2019 Route: dc30ef2ad294a732|2022-04-05--09-45-30
mock: dc30ef2ad294a732|2022-04-05--08-31-18

### What hardware does this issue affect?

comma three

### Which car does this affect?

Subaru Impreza 2017

### Provide a route where the issue occurs

dc30ef2ad294a732|2022-04-05--09-45-30

### openpilot version

b2b6205e252b5291a953dd94c5524715b1f8f444

sshane, 2022-04-05:
Some of your fingerprints don't exist in values.py, so you're fuzzy fingerprinting. Not sure about how deterministic fuzzy fingerprinting is, but let's get you fingerprinted!

michaelhonan, 2022-04-05:
@martinl have you seen anything like this in your FPv2 testing?

martinl, 2022-04-05:
There are still Impreza 2020 and Forester FPv1 fingerprints in master, so I would assume Forester one is actually FPv1 (mis)match. Should be easy to verify using fingerprintSource in carParams log.

michaelhonan, 2022-04-05:
I think you're onto it, the `SUBARU FORESTER 2019` one matched with `'fingerprintSource': 'can'`.

Seems the solution is to add my FW entries then. Doesn't explain the weird `esp` fw flipping, but I'll see if the changes work on my fork before creating a PR.

sshane, 2022-04-12:
@michaelhonan see if this works for you: https://github.com/commaai/openpilot/pull/24194

adeebshihadeh, 2022-04-12:
Also looks like we'll have to delete the remaining CAN fingerprints #24195. Fortunately, not too many users are still fingerprinting with CAN.

michaelhonan, 2022-04-12:
@sshane Worked a treat mate, even with the CAN fingerprints removed.

closed 2022-04-13 by sshane