commaai/openpilot #24134: Randomly stop working

Deivis1298, 2022-04-05:
### Describe the bug

The coma 3 works but it randomly stopped working and it trip a code on the car 
#rav42017

The code are random 
Either blind spot monitor not working
Radar not working 
A steering control not working

4436008557 

I never had a code before installing it on my car , i resenty updated it to 8.13

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

It’s randomly 

### openpilot version

0.8.13

pd0wm, 2022-04-05:
This sounds like a hardware issue. Please contact support@comma.ai. Without any routes or dongle ids we can't look into it.

closed 2022-04-05 by pd0wm

Deivis1298, 2022-04-05:
How can i provide you want u need