commaai/openpilot #24135: Cruise Fault immediately after releasing the accelerator when engaged.

Verylukyguy, 2022-04-05:
### Describe the bug

When OpenPilot is already engaged and you begin accelerating with the Disengage on Accelerator toggle turned off, you get a Cruise Fault: Take Control Immediately RSOD after releasing the accelerator.

This also happens when you engage while accelerating and the accelerator is released.

I provided a route with several examples.

### What hardware does this issue affect?

comma two

### Provide a route where the issue occurs

e38a02cae1ef1e0c|2022-04-05--11-10-35--0

### openpilot version

Comma Master 0.8.14 @ f0ac808

sshane, 2022-04-05:
Can you try this branch out? https://github.com/commaai/openpilot/tree/gm-cruise-faults

I think it may be related to how we now use `longActive` instead of `enabled` for a couple of signals related to ACC (changed in PR https://github.com/commaai/openpilot/pull/23859).

When you press the gas, we set `ASCMGasRegenCmd->GasRegenCmdActive` to False which causes `AcceleratorPedal2->CruiseState` to fall, then when you release the gas we start commanding gas/regen immediately while the CruiseState takes a frame to rise again, so maybe that's the cause of the faults? Let me know how the branch goes! @Verylukyguy

[image]

Verylukyguy, 2022-04-06:
> Can you try this branch out? https://github.com/commaai/openpilot/tree/gm-cruise-faults
>
> I think it may be related to how we now use `longActive` instead of `enabled` for a couple of signals related [...]
This made it worse; now it faults when the accelerator is pressed while engaged.

sshane, 2022-04-06:
It looks like this might have always been an issue then. We probably need a stock route of an ICE car to see what the ASCM sends to avoid faults. Doesn't seem to be an issue on the Volts, which is what @twilsonco said the GM port started with.

Verylukyguy, 2022-04-06:
It has been an ongoing issue on stock OpenPilot, but it is very strange that the Volt code didn't work, because I am usually on some version of Tim's fork and I do not get those errors.  I will attempt to get you some factory GM routes to analyze.

Verylukyguy, 2022-04-06:
I think that I actually found a reference to this issue in https://github.com/commaai/openpilot/pull/348

twilsonco, 2022-04-07:
This happens in my '18 Volt with C3 as well.

steps to reproduce on apparently any GM vehicle:
1) in master, disable the disengage on gas toggle
2) Drive, engage
3) Press gas; lateral stays engaged, but you'll have long control with the gas pedal
4) release gas pedal

Expected behavior: OP resumes long control
Observed behavior: RSOD "Cruise Faulted Take control immediately". This resolves within seconds and you can reengage.

twilsonco, 2022-04-07:
I'm stumped for the time being. Reaching out to @JasonJShuler and @qadmus for help. Here's from discord:

> steering stays engaged while you press the gas, but then ACC faults right after you release the pedal. In these plots I got from Shane of stock GM cruise doing it's no-disengage-on-gas thing, I notice that, at t=148s when gasPressed goes to 1, the acc keeps sending a non-"zero" GasRegenCmd for ~1s before it drops to "zero", then it comes back to the 1900 value ~0.5s before the gas is released.
> 
> I thought maybe this meant there had to be a non-"zero" gas command present when op-long resumes control, but I don't think that makes sense now...
> More likely, we're missing a frame in the ACC commands in one of two ways. When the user releases the gas,  op sends the first true GasRegenCmdActive signal either:
> 1) a frame too early, overlapping with the gas pedal and causing a fault, or
> 2) a frame too late, resulting in missing frames and causing a fault.
> Importantly, when this was thought to be confined to Acadia, @Shane indicated that it it's not resolved Acadia would be on the dashcam list. Given the L&P harness and other GM shenanigans, I'm concerned they might ax all of GM.

[image]

closed 2022-04-08 by sshane