commaai/openpilot #24146: Sticking brake command for 2019 Toyota Sienna with SDSU

cbom32, 2022-04-06:
### Describe the bug

openpilot allows both gas and brake commands to be applied with a SDSU. Initially when disengaging openpilot on gas pedal with openpilot slowing down, the brake command will stick (further tests reveal other methods bug shows up). When gas is applied, the brake will still be engaged and will not release until car is turned off for about 30 seconds. The force of the brake applied seems to be the "last brake" command...as if I'm going 55 mph and openpilot decides to go 45 mph (car in front), the brake command to get to 45 mph will maintain, so when the car tries to get back to 55 mph, openpilot has to overcome the brake command...eventually resulting in overheating on the brakes. As the car slows down further, through manual intervention (openpilot disengaged), the brake command still sticks and gets worse until car is turned off. 

Occurs on both stock openpilot 0.8.13 (Comma 2) and 0.8.14 Master-ci big model (Comma 3). First occurrence after openpilot engaged, with brake command being maintained with openpilot engaged and maintained after openpilot disengaged. Occurs on disengagement via user brake press, user gas press, manual cruse control turned off. 

Next steps for testing will be to keep SDSU disconnected and use a fork to prove the stuck brake issue is from SDSU (have to use fork due to fp v1 on Toyota with disconnected DSU).

### What hardware does this issue affect?

comma three, comma two

### Which car does this affect?

Toyota Sienna 2019

### Provide a route where the issue occurs

7cbeaba98d732562|2022-04-05--14-16-08--0

### openpilot version

0.8.13, 0.8.14 Master-ci big model

### Additional info

- Route after the issue without OpenPilot engaged...issue still happens: 7cbeaba98d732562|2022-04-05--14-22-03--1
- Route after disconnecting SDSU: 7cbeaba98d732562|2022-04-05--14-28-42--1
- First captured occurrence: ad9d2279abbc8220c|2022-03-21--18-13-21--4

sshane, 2022-04-06:
Hi there! What is the issue with fingerprinting on stock openpilot? You seem to have fingerprinting 2.0 and when I ran your CarParams through fingerprinting, it detected your car with/without the dsu fw.

And yes, try to test in the exact same situations without the smartDsu and see if the brake pressure issue persists.

cbom32, 2022-04-06:
Good call. fp v1 issue I had was in 0.8.11 and at the time SDSU was purchased and resolved. Been awhile running op with unplugged DSU. Stock 0.8.13 fp v2 works, so thank you!

Unplugged SDSU: So far with limited testing haven't had the stuck brake issue on either stock 0.8.13 on Comma 2 or sunnyhaibin/0.8.12-prod-full-toyota. I'll stick with testing on stock and report back after more complete testing.

cbom32, 2022-04-07:
Can confirm same issue occurs with SDSU disconnected. So SDSU not the cause. I'll update with affected route, which is exactly same situation as the very first occurrence. 

Should I try an older op rev, like 0.8.10 to confirm it is a recent bug?

sshane, 2022-04-07:
From that route, it looks like the brake pressure drops to zero for quite some time, despite openpilot commanded a lot of brake, and the car actually braking. Then when you disengage, the brake pressure shoots up to ~50 and slowly rises to ~60 over the next 30 seconds.

Can you try to disable openpilot (`Enable openpilot` toggled off in settings) and use the car's stock cruise control system to see if it does the same thing?

[image]

abdulrahman004, 2022-04-07:
Following as a Seinna 2018 user with SDSU. If the brake pressure is 0 and Brake_pressed i 0, how is the car even slowing Down ?

cydia2020, 2022-04-10:
> Following as a Seinna 2018 user with SDSU. If the brake pressure is 0 and Brake_pressed i 0, how is the car even slowing Down ?

Engine brake, might be helpful to look at engine RPM as well.

cydia2020, 2022-04-10:
Since we're always sending ACC control, it might be helpful to set `PERMIT_BRAKING` to 0 when the system is off, I think stock does this, this signal is quite important in some other unrelated issues as well.

cbom32, 2022-04-10:
Update: Had brakes and brake system checked out at Toyota dealership. They ran diagnostics, drove 10 miles to try and reproduce issue. Couldn't reproduce. I did put everything back to factory condition before they ran their tests: removed C2/C3, removed car harness, OBDII/RJ-45 plug in, etc. No obvious signs of damaged components to brake system. They did see wear on the front brake pads and slightly warped router...results of the sticking brake issue. 

I've been driving the last day as factory. Getting a new recommend USB-C cable to try and make sure no cable issues. 

Any other tests I should try? I haven't yet tried without Openpilot toggle, so I'll do that and report again.

sshane, 2022-04-23:
I checked a few hundred thousand segments available here from the past 60 days, and didn't see anything similar to what you were seeing in our data. I ignored the RAV4 and Corolla which are notorious for having noisy brake pressure signals that are routinely non-zero when brake isn't pressed (only has steps in 256 from 0 to 4096). The two checks were if the brake pressure was high when the gas is pressed, and we don't expect any braking recently, OR if the brake pressure signal is abnormally noisy.

```
Total segments checked: 994710
By platform:
TOYOTA PRIUS 2017               - 165726 segments
TOYOTA RAV4 HYBRID 2019         - 121776 segments
TOYOTA COROLLA TSS2 2019        - 85372 segments
TOYOTA RAV4 2019                - 79761 segments
TOYOTA CAMRY 2018               - 63378 segments
TOYOTA COROLLA HYBRID TSS2 2019 - 58029 segments
TOYOTA PRIUS TSS2 2021          - 53292 segments
TOYOTA HIGHLANDER HYBRID 2020   - 42935 segments
TOYOTA HIGHLANDER 2017          - 37645 segments
TOYOTA RAV4 2017                - 29619 segments
TOYOTA RAV4 HYBRID 2017         - 29244 segments
TOYOTA SIENNA 2018              - 28469 segments
TOYOTA HIGHLANDER 2020          - 28246 segments
TOYOTA COROLLA 2017             - 27073 segments
TOYOTA CAMRY HYBRID 2018        - 24453 segments
LEXUS RX 2016                   - 19192 segments
TOYOTA CAMRY HYBRID 2021        - 16677 segments
TOYOTA HIGHLANDER HYBRID 2018   - 14410 segments
LEXUS RX HYBRID 2017            - 11551 segments
TOYOTA CAMRY 2021               - 11442 segments
LEXUS NX HYBRID 2018            - 5775 segments
LEXUS ES HYBRID 2019            - 5471 segments
LEXUS ES 2019                   - 5029 segments
LEXUS NX 2018                   - 4298 segments
TOYOTA C-HR 2018                - 4064 segments
TOYOTA AVALON 2019              - 3385 segments
LEXUS IS 2018                   - 3312 segments
TOYOTA MIRAI 2021               - 3166 segments
TOYOTA AVALON HYBRID 2019       - 2823 segments
LEXUS NX 2020                   - 2585 segments
LEXUS RX 2020                   - 2558 segments
LEXUS RX HYBRID 2020            - 1248 segments
TOYOTA AVALON 2016              - 998 segments
TOYOTA C-HR HYBRID 2018         - 920 segments
TOYOTA AVALON 2022              - 463 segments
TOYOTA ALPHARD 2020             - 219 segments
LEXUS ES HYBRID 2018            - 105 segments
TOYOTA PRIUS v 2017             - 1 segments

Problem segments:
TOYOTA HIGHLANDER 2020          - frames above brake pressure threshold: 108, noisiness: 40
TOYOTA HIGHLANDER 2020          - frames above brake pressure threshold: 1198, noisiness: 1
TOYOTA HIGHLANDER 2020          - frames above brake pressure threshold: 1198, noisiness: 0
TOYOTA HIGHLANDER 2020          - frames above brake pressure threshold: 211, noisiness: 55
TOYOTA HIGHLANDER 2020          - frames above brake pressure threshold: 2325, noisiness: 36
TOYOTA HIGHLANDER 2020          - frames above brake pressure threshold: 40, noisiness: 7
TOYOTA HIGHLANDER 2020          - frames above brake pressure threshold: 1064, noisiness: 66
TOYOTA HIGHLANDER 2020          - frames above brake pressure threshold: 1260, noisiness: 2
TOYOTA HIGHLANDER 2020          - frames above brake pressure threshold: 192, noisiness: 19
TOYOTA HIGHLANDER 2020          - frames above brake pressure threshold: 667, noisiness: 63
TOYOTA HIGHLANDER 2020          - frames above brake pressure threshold: 480, noisiness: 78
TOYOTA SIENNA 2018              - frames above brake pressure threshold: 1685, noisiness: 524
TOYOTA SIENNA 2018              - frames above brake pressure threshold: 1308, noisiness: 516
TOYOTA SIENNA 2018              - frames above brake pressure threshold: 0, noisiness: 1826
TOYOTA SIENNA 2018              - frames above brake pressure threshold: 0, noisiness: 879
```

I checked all 309 segments that we have rlogs for from your dongle ID manually and there were only 5 segments with the issue. It only detected four here as I wrote the detector to check that the gas is pressed to reduce the false positives, as on the Sienna (and possibly others?) the brake pressed bit in `BRAKE_MODULE` can sometimes always be zero when disengaged, even if the user is braking and brake pressure is high. Needing to see the gas being pressed, along with a high brake pressure nets pretty confident results.

All the Highlander segments here are just weird, [the timing from the openpilot messages (carState, etc.) to the CAN messages is just completely wrong](https://user-images.githubusercontent.com/25857203/164874709-9b4a9f4c-7894-4a3d-acaf-57664b4a8385.png), so it thinks the brake is being pressed when it's not, so those can be ignored.

If you get the time, try driving around for an extended period of time with openpilot toggled off in settings so it's just recording data. If you see any issues there it sounds like a rare car-specific bug.

sshane, 2022-04-23:
Side question, at the dealer did they test with the cruise control, or only manually driving? Did you notice this happening when **not** using cruise control? Perhaps that's why they didn't find anything.

sshane, 2022-06-02:
@cbom32 any updates to this issue?

sshane, 2022-07-07:
@cbom32 Is this still an issue?

cbom32, 2022-07-07:
No longer an issue after long term testing on 0.8.14 master-ci and after release of 0.8.14. Got back from a 700 mile round trip with no issues. Two things to my understanding have resolved/prevented the issue: 1. No disengage on gas pedal introduced in 0.8.14 master-ci. 2. Adding extension to USB-C cable (original cable was pinched...placement a known issue for Siennas). 

Thank you for your help in researching for other instances and following up!