commaai/openpilot #24171: Is there anyway to get a MAPBOX_TOKEN for pc?

unomove, 2022-04-08:
### Describe the bug

When I debug the map_renderer.py. I got the following error message:
```
[ INFO ]  "{unknown}[General]: GPU Identifier: NVIDIA GeForce RTX 2080 Ti/PCIe/SSE2"
[ ERROR ]  "{unknown}[Setup]: loading style failed: HTTP status code 403"
```
However, when I use "https://api.commadotai.com/#comma-api-spec"  to get a jwt token.
```
export MAPBOX_TOKEN=eyJ0eXxxxxxxxxxxxxxxxxxxxxxx
```
I got the 401 error.
```
[ INFO ]  "{unknown}[General]: GPU Identifier: NVIDIA GeForce RTX 2080 Ti/PCIe/SSE2"
[ ERROR ]  "{unknown}[Setup]: loading style failed: HTTP status code 401"
```
Then, If I use my own MAPBOX token from "https://account.mapbox.com/". I got the following error.
```
[ INFO ]  "{unknown}[General]: GPU Identifier: NVIDIA GeForce RTX 2080 Ti/PCIe/SSE2"
[ ERROR ]  "{unknown}[General]: The resource `https://api.mapbox.com/styles/v1/commaai/ckvmksrpd4n0a14pfdo5heqzr` not found"
[ ERROR ]  "{unknown}[Setup]: loading style failed: HTTP status code 404"
```

It seems that the error comes from the MAPBOX_TOKEN issued from comma.ai. Is there anyway to debug/test the function without buying a comma three device?

### OS Version

Ubuntu 20.04

adeebshihadeh, 2022-04-08:
The token is from Mapbox, not us.

closed 2022-04-08 by adeebshihadeh

unomove, 2022-04-09:
No. The problem is not the MAPBOX-token. If I use my own token. I cann't access the style files of comma. How do I deal with it?

See my previous comments.
```
[ INFO ]  "{unknown}[General]: GPU Identifier: NVIDIA GeForce RTX 2080 Ti/PCIe/SSE2"
[ ERROR ]  "{unknown}[General]: The resource `https://api.mapbox.com/styles/v1/commaai/ckvmksrpd4n0a14pfdo5heqzr` not found"
[ ERROR ]  "{unknown}[Setup]: loading style failed: HTTP status code 404"
```