commaai/openpilot #24172: Civic controls mismatch

adeebshihadeh, 2022-04-08:
3x controls mismatches across these routes

cfb32f0fb91b173b|2022-04-06--14-54-45
cfb32f0fb91b173b|2022-04-08--08-38-33

closed 2022-04-12 by sshane

redacid95, 2022-04-13:
20' Civic Hatchback Sport. Still getting control mismatches, less than what I was getting before this update. Disable on accel is set to false, and VOACC (OP Long) is enabled.

f1055ddb93ff2d3b|2022-04-12--17-30-13--4
f1055ddb93ff2d3b|2022-04-12--17-30-13--5
f1055ddb93ff2d3b|2022-04-12--17-30-13--6

adeebshihadeh, 2022-04-13:
This is specifically for cars with a gas interceptor.

sshane, 2022-04-13:
@redacid95 controls allowed didn't go high when you engaged, does this happen with stock ACC or only openpilot longitudinal? In any case you can open an issue and I'll look into it when I have time