commaai/openpilot #24174: Cruise faults when using VOACC on Honda Bosch

joelspiers15, 2022-04-08:
### Describe the bug

To reproduce:

1. Install e2e_long branch
2. Set the DisableRadar_Allow  param to 1
3. Enable OP long (VOACC)
4. Attempt to engage a few times

It would usually fault on the second attempt to engage but not always. If you need more examples other than the route I provided I have a few.

### What hardware does this issue affect?

comma three

### Which car does this affect?

Honda Accord 2018

### Provide a route where the issue occurs

3bac110059677809|2022-04-08--16-44-59--0

### openpilot version

45c0ea754b6aa571a30de8a58a3070e757786c77

briantran33, 2022-04-09:
This happened to me as well. I have a 2019 Honda Accord 2019. Comma 3 as well running the latest version of e2e_long.

redacid95, 2022-04-13:
Here are the 3 segments from my related issue. 

f1055ddb93ff2d3b|2022-04-12--17-30-13--4
f1055ddb93ff2d3b|2022-04-12--17-30-13--5
f1055ddb93ff2d3b|2022-04-12--17-30-13--6

kjake, 2022-04-14:
Also happens in a 2017 CR-V.

redacid95, 2022-04-27:
Is this going to be fixed before 8.14 release? Does anyone have a recommendation for me to look at to fix this issue? This is an issue in master not just e2e_long to be clear.

redacid95, 2022-05-28:
So after the above merge request, there is now some new behavior to report. I can get OP to engage if I pressing on the gas pedal, engage, and then release the gas pedal. Every 2 or so engagements, I get a controls mismatch error while pressing the gas pedal, but then I am able to get it to engage immediately after while still pressing the gas. If I am not pressing the gas when I engage, it immediately cruise faults.

joelspiers15, 2022-06-02:
@redacid95 What commit did you get that behavior on? I tried `cherry-pick`ing the typo PR into the latest e2e_long and couldn't get any engagements without the mismatch.

redacid95, 2022-06-02:
@joelspiers15 this is happening on 0.8.14 not just e2e_long. I am on the latest master with this problem. I haven't tested e2e_long in months.

joelspiers15, 2022-06-02:
@redacid95 Tried on the latest master ( c677c6b164fc8e642ca75809213eca0678cf86ac ) as well which would show "waiting for controls to start" when I started the car with VOACC enabled. Maybe it's master 4 days ago I should try to get that behavior from.

redacid95, 2022-06-02:
@joelspiers15  oh no. Sounds like there might be a new issue happening. I haven't updated in about a week. I haven't used my C3 since this issue has started except to test fixes... VOACC is like the only reason why I use my comma. I hate stock long.

redacid95, 2022-06-02:
Can confirm “waiting for controls to start”. Wasn’t like that a week ago. I’ll see if I can find what the cause is or submit an issue.

redacid95, 2022-06-02:
Found the controls start issue to be with commit #24311. Unrelated to this issue.

Vaggysag, 2022-06-08:
I fixed this issue by reverting panda commit
https://github.com/commaai/panda/commit/4acaebb62bbb5b7d77ef984b73ece6940cc1d28a

closed 2022-06-09 by sshane