commaai/openpilot #24189: camerad: Use the ISP for debayering, save CPU / GPU / ram bandwidth.

geohot, 2022-04-11:
Parts 1 and 2 of #24187 are prereqs for this. The ISP has a custom mini program to do things. Might be possible to get from Pixel 3.

See [camera_fixups_isp](https://github.com/commaai/openpilot/tree/camera_fixups_isp) for some work toward this.

There's two ISPs, so the dcam will have to be a little late, which should be okay. I suspect these ISPs can debayer super fast.

geohot, 2022-04-26:
Would offer a $2,000 bounty for this.

jon-chuang, 2023-02-14:
Hi, is this still open? Could I be assigned?

Sl4y3rJ3ns, 2024-09-18:
Is this bounty still open or has Comma taken this into their own hands? I'm willing to work on this bounty because I have a reference Pixel 3 available.

closed 2024-11-08 by adeebshihadeh