commaai/openpilot #24192: crash in onnx_runner (modeld)

ebadi, 2022-04-12:
### Describe the bug

Hi,
After building the docker images locally:
```
controlsState: Reader was evicted, reconnecting
Onnx available providers:  ['TensorrtExecutionProvider', 'CUDAExecutionProvider', 'CPUExecutionProvider']
Onnx selected provider:  ['CUDAExecutionProvider']
clocksd logcatd modeld navd proclogd ui soundd locationd calibrationd controlsd deleter logmessaged paramsd plannerd radard thermald statsd
2022-04-12 10:47:16.707337515 [W:onnxruntime:Default, onnxruntime_pybind_state.cc:552 CreateExecutionProviderInstance] Failed to create CUDAExecutionProvider. Please reference https://onnxruntime.ai/docs/reference/execution-providers/CUDA-ExecutionProvider.html#requirements to ensure all dependencies are met.
Onnx using  ['CPUExecutionProvider']
ready to run onnx model ['input_imgs', 'big_input_imgs', 'desire', 'traffic_convention', 'initial_state'] [[1, 12, 128, 256], [1, 12, 128, 256], [1, 8], [1, 2], [1, 512]]
Traceback (most recent call last):
  File "/openpilot/selfdrive/modeld/runners/onnx_runner.py", line 60, in <module>
    run_loop(ort_session)
  File "/openpilot/selfdrive/modeld/runners/onnx_runner.py", line 34, in run_loop
    inputs.append(read(ts).reshape(shp))
  File "/openpilot/selfdrive/modeld/runners/onnx_runner.py", line 17, in read
    assert(len(st) > 0)
AssertionError
```

### OS Version

Ubuntu 18.04

### openpilot version or commit

85c714e350b2445a78299fe6df8c06a36d80bcea

adeebshihadeh, 2022-04-12:
duplicate of #23666

closed 2022-04-12 by adeebshihadeh

ebadi, 2022-04-13:
@adeebshihadeh  Thanks. I am aware of that issue. 
I had the same issue [here](https://github.com/commaai/openpilot/issues/23985#issuecomment-1078318406) that was resolved by the pull request mentioned [here](https://github.com/commaai/openpilot/issues/23985#issuecomment-1078770530). My guess is that changes after https://github.com/commaai/openpilot/commit/d014d8191d950a56a961c18c84463ffd0d3d5de5 caused the bug to appear again.