commaai/openpilot #24199: Honda Bosch: Controls Mismatch

redacid95, 2022-04-13:
### Describe the bug

Getting control mismatches, less than what I was getting before this update #24173. Disable on accel is set to false, and VOACC (OP Long) is enabled.

I was able to successfully engage a few times before these segments. I believe it may be something to do with maintaining lateral control while gas is pressed, but I was able to successfully before these mismatch alerts.

f1055ddb93ff2d3b|2022-04-12--17-30-13--4
f1055ddb93ff2d3b|2022-04-12--17-30-13--5
f1055ddb93ff2d3b|2022-04-12--17-30-13--6

### What hardware does this issue affect?

comma three

### Which car does this affect?

20' Civic Hatchback Sport

### Provide a route where the issue occurs

f1055ddb93ff2d3b|2022-04-12--17-30-13--4

### openpilot version

master

redacid95, 2022-04-13:
@sshane In answer to your question in the other issue log - I just tested on stock ACC, and was not able to replicate.

redacid95, 2022-04-13:
Getting cruise faults as well. While I'm not on e2e_long, it might be related to #24174. I believe the last of the 3 segments I sent before was also a cruise fault. On VOACC. I'll get latest segment soon.

pd0wm, 2022-04-13:
@sshane Duplicate of https://github.com/commaai/openpilot/issues/24174?

closed 2022-04-13 by sshane

redacid95, 2022-04-13:
So this is a duplicate? I know they are reporting on e2e_long and only about cruise faults. I'm getting control mismatch not just cruise faults and on master branch.

pd0wm, 2022-04-13:
Cruise faults are usually caused by the panda blocking messages, which is caused by a controls mismatch. Both issues most likely have the same underlying cause.

redacid95, 2022-04-13:
Thank you for the explanation. If I had the time I'd dive in and learn more about this stuff.