commaai/openpilot #24218: Very slow acceleration when resuming from a stop or slow roll.

Verylukyguy, 2022-04-15:
### Describe the bug

Very slow acceleration when resuming from a stop or after slowing to a near stop for a lead car to turn; it does get normal at around 30 mph.

### What hardware does this issue affect?

comma two

### Which car does this affect?

GMC Acadia 2018

### Provide a route where the issue occurs

e38a02cae1ef1e0c|2022-04-15--07-43-35--3

### openpilot version

Master 0.8.14 @ 1fa1c44

### Additional info

This has been an ongoing issue from the beginning for the GM port.

@Twilsonco has fixed this in his fork and has even added a toggle for Normal/Sport/Eco/Creep acceleration modes.

twilsonco, 2022-04-15:
This is due to [MAX\_GAS](https://github.com/commaai/openpilot/blob/1fa1c44342995d27ed9295429ac4a0506c7569a6/selfdrive/car/gm/values.py#L26) (and it's [corresponding value in panda](https://github.com/commaai/panda/blob/c925407461e46adf3282494af6daa00b2626ebb8/board/safety/safety_gm.h#L18))in selfdrive/car/gm/values being set too low. The max value for the gm gas command is 4095, which is what I use for max gas in my fork. The actual acceleration is then limited by the acceleration limits in OP. I think 4095 is excessive if you don't have a "sport acceleration" mode, but a higher value such as 3584 (half-way from the current value to the max) will allow for more reasonable acceleration (still less than stock ACC is capable of).

Verylukyguy, 2022-10-17:
Here is another route where the lead pulls away and the vehicle has very slow acceleration when resuming.

df51b2195de66a9b|2022-10-16--15-55-34--13

Verylukyguy, 2023-02-08:
Here is a new route on Comma Master at 5b9f2c3

df51b2195de66a9b|2023-02-08--10-02-10--5

sshane, 2023-08-07:
@Verylukyguy if you can rally the other ICE GMs to gather routes from a stop, that will help us raise the ICE gas/regen limit across the board.

1454, 2023-08-08:
Comma 3, silverado here. Same issue. Excessively slow take up acceleration until about 30ish. Usually have to manually override.

Verylukyguy, 2023-08-08:
Here is a stock ACC route in Dashcam Mode:  df51b2195de66a9b|2023-08-08--16-37-24--9
At this point in the route, my lead turns off and I experience the fastest acceleration in the route.
I let the stock ACC do most of the acceleration on this entire route.

sshane, 2023-08-08:
@1454 please provide a route with openpilot toggled off in the Device settings, and have the car come from a stop to at least 30 mph on flat ground. This will help us standardize a solution for all platforms!

sshane, 2023-08-08:
@Verylukyguy anything from closer to a stop?

Verylukyguy, 2023-08-08:
There is a factory ACC resume from a stop at **16:43:29 – 6**

sshane, 2023-08-08:
From cookiemonster: fe80a4e1cedec853|2023-08-08--08-58-41

morrislee, 2023-08-09:
stock acc route with come to a stop and start from standing still fe80a4e1cedec853|2023-08-08--18-10-55--12

Verylukyguy, 2023-08-11:
Here is another stock ACC route with a few examples of accelerations from a stop 
df51b2195de66a9b|2023-08-11--17-19-41--7

1454, 2023-08-13:
I have completed a 4-30+, and 0-27 drives. Traffic slowed up on the second one before I hit 30.

61c6258cac78af08|2023-08-13--13-07-56--0

Verylukyguy, 2023-09-12:
Referencing Shane's newest discovery: https://github.com/commaai/opendbc/pull/951