commaai/openpilot #24232: “Openpilot unavaiable  communication issue between processes" issue on latest OP container version

fatsheep2020, 2022-04-17:
### Describe the bug

run carla and OP container in PC , HMI show error: "Openpilot unavaiable "communication issue between processes"
when attempt to key in "1"

How to reproduce:
  sudo INSTALL=1 ./start_carla.sh
   sudo ./start_openpilot_docker.sh
  autopilot should enable when press key "1"
Expected behavior:
   expected to be able to dive like the video
  https://www.youtube.com/watch?v=lPQHv3z3MWY
Actual behavior:
  HMI show error: "Openpilot unavaiable "communication issue between processes"
  vehicle not move

no error logs

openpilot container info:
   ghcr.io/commaai/openpilot-sim:latest

openpilot version :  v0.8.14

### OS Version

Ubuntu 20.04

### openpilot version or commit

v0.8.14

pd0wm, 2022-04-20:
Sounds like a duplicate of https://github.com/commaai/openpilot/issues/23666. Closing due lack of logs.

closed 2022-04-20 by pd0wm

fatsheep2020, 2022-05-08:
@pd0wm  add log info after some deep [analysis:
here is the error part when run launch_openpilot.sh in openpilot container:
[image]

it seems like error happen when run the model supercombo.onnx in onnx_runner.py, at same time, check the file, it exist:
[image]

thanks!

fatsheep2020, 2022-05-08:
@pd0wm  one more question: do you think we should use this onnx model?
thanks!