commaai/openpilot #24236: Resume route when coming back onroad

LoganxDev, 2022-04-17:
If route has not been completed (Maybe check within a limit of so many yards to see if route is complete. Not sure how this is done today? Probably just ends on offroad) cache route on offroad, then grab route when returning to onroad and ask to resume.

LoganxDev, 2022-04-17:
- [ ] Cache route if route not finished and entering offroad
- [ ] Grab cached route returning to onroad
- [ ] Add UI for resume

pd0wm, 2022-04-20:
This is one of the mocks we drew up for this feature when we were making the initial design for nav.

[image]

LoganxDev, 2022-04-20:
That route really lends itself nicely to this design. If the route were mirrored over the y axis the dialog box would be covering the route unless the map was further zoomed out. I wonder if having the dialog be a tad shorter and spanning the bottom of the map window would work better. 

I also don't see a clear way for the user to cancel the route.

pd0wm, 2022-04-20:
Making the dialog wider and shorter is fine with me. Alternatively you can figure out how to rotate the map to make it fit, but that might look confusing if you need more than 45 degrees of rotation.

The plan was to make the dialog go away ~10 seconds after you start moving, or when you close the map window.

adeebshihadeh, 2023-06-25:
I think the new nav entry on the home screen and in the map should cover this use case pretty well.

closed 2023-06-25 by adeebshihadeh