commaai/openpilot #24277: Make mypy checks stricter

adeebshihadeh, 2022-04-20:
I suggest enabling each flag and fixing the violating in separate PRs.

Looked over the docs a bit and think these are the flags we want to enable. Perhaps there's some more we want or some in this list we don't want.
```
diff --git a/.pre-commit-config.yaml b/.pre-commit-config.yaml
index 3fc559656..c077ea8e7 100644
--- a/.pre-commit-config.yaml
+++ b/.pre-commit-config.yaml
@@ -22,6 +22,12 @@ repos:
         exclude: '^(pyextra/)|(cereal/)|(rednose/)|(panda/)|(laika/)|(opendbc/)|(laika_repo/)|(rednose_repo/)/'
         additional_dependencies: ['git+https://github.com/numpy/numpy-stubs', 'types-requests', 'types-atomicwrites',
                                   'types-pycurl']
+        args:
+        - --check-untyped-defs
+        - --warn-redundant-casts
+        - --warn-unused-ignores
+        - --warn-return-any
+        - --warn-unreachable
```

herman5, 2022-04-30:
- [x] --warn-redundant-casts #24304
- [x] --warn-unused-ignores #24287
- [x] --warn-unreachable
- [ ] --warn-return-any
 - #24284
 - #24379

adeebshihadeh, 2022-05-02:
It would also be nice if there was some way to verify that our coverage isn't going down over time. It seems like subtle changes can drastically affect the mypy coverage silently.

herman5, 2022-05-13:
> some way to verify that our coverage isn't going down over time. It seems like subtle changes can drastically affect the mypy coverage silently.

Now that #24379 has landed, we could track that in a separate issue and close this one out.

closed 2022-05-23 by pd0wm