commaai/openpilot #24305: C3 video stream freezes but everything else works

royjr, 2022-04-24:
### Describe the bug

Video will freeze randomly while OP active, UI is still updated and OP driving normal. Reboot fixes it.

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

1a5d045d2c531a6d|2022-04-23--21-47-31

### openpilot version

0.8.14

### Additional info

Simple fork off master with added support for civic. Just merged last couple weeks of updates so probably something committed recently. Another discord user reported same issue today as well.

KyleGreenlaw, 2022-04-24:
Having the same issues. Video stream freezes, updates a few seconds later for me

geohot, 2022-04-24:
Should be fixed in master with YUV revert.

closed 2022-04-24 by geohot