commaai/openpilot #24316: 2021 Hyunda Palisade - Car Alert - Check Forward Collision Avoidance Assist System

llck80ll, 2022-04-25:
### Describe the bug

When Comma 3 is plugged in, whether OpenPilot is engaged or not, the alert keeps coming up. I tried rebooting, unplugging but doesn't seem to correct the issue.

### What hardware does this issue affect?

comma three

### Which car does this affect?

Hyundai Palisade 2021

### Provide a route where the issue occurs

29c3af39bd48a377|2022-04-25--07-52-41--0

### openpilot version

0.8.13

### Additional info

It's been happening for the past 3-4 days or so.

sshane, 2022-04-26:
Can you connect your device to WiFi to let the logs upload?

llck80ll, 2022-04-26:
I apologize, you’re right. The device is in the center console since the alert keeps going off. I’ll plug it in inside the house so it connects to Wi-Fi now.

llck80ll, 2022-05-19:
Hello, any updates on this car alert? I’m going to try uninstalling and reinstalling OpenPilot.

It is connected to Wi-Fi.

llck80ll, 2022-05-19:
I still get the alert after doing a factory reset.

llck80ll, 2022-10-11:
My car is parked in the garage and is connected to Wi-Fi at all times.

Is there a way to manually push the logs?

llck80ll, 2022-10-11:
It is connected to Wi-Fi.

sshane, 2023-05-10:
@llck80ll I noticed you haven't driven in a while and I wasn't able to preserve any routes, do you still have this issue?

llck80ll, 2023-05-10:
Palisades is totaled.

closed 2023-08-07 by sshane