commaai/openpilot #24323: Openpilot does not drive in CARLA simulator

ebadi, 2022-04-25:
### Describe the bug

It is no longer possible to run the simulator & openpilot together and no image appears in openpilot window. 

```
./launch_openpilot.sh
navd ui soundd deleter logmessaged thermald statsd
```
```
./bridge.py
frame:  engaged: True ; throttle:  0.0 ; steer(c/deg):  -0.0 0.0 ; brake:  0.0
```
My guess is that the change in #24320 caused this.  

### OS Version

Ubuntu 18.04

### openpilot version or commit

01968ca61ba32174f86f826c5b4166890034691c

ebadi, 2022-04-25:
I think I identified the root cause of the issue. I see many errors in this format when I stop carla docker:

```
ERROR: Invalid session: no stream available with id 9 
ERROR: Invalid session: no stream available with id 2 
ERROR: Invalid session: no stream available with id 8 
ERROR: Invalid session: no stream available with id 6 
ERROR: Invalid session: no stream available with id 7 
ERROR: Invalid session: no stream available with id 4 
ERROR: Invalid session: no stream available with id 10 
ERROR: Invalid session: no stream available with id 9 
ERROR: Invalid session: no stream available with id 5 
ERROR: Invalid session: no stream available with id 8 

```

maykonpacheco, 2022-04-26:
I have the same problem.

maykonpacheco, 2022-04-26:
[image]

ebadi, 2022-04-26:
> [image]

I think you have to press `1` to engage openpilot (and I did it) but it is not driving.
I also think that is important to remove old docker images to reproduce the issue.

maykonpacheco, 2022-04-26:
That's exactly what I did, delete Everything that was old and installed it again.  Now when I press 1 to start this error happened.

maykonpacheco, 2022-04-26:
It started working again, but the car can't make any turns

ebadi, 2022-04-27:
Recent changes fixed this for me. Not sure if I have to keep this one open.

pd0wm, 2022-04-28:
Looks like there is a lag between the video feed from the simulator and the current state, causing it to drive very poorly. We'll look into it.

closed 2022-04-28 by pd0wm

ebadi, 2022-04-29:
thanks @pd0wm, it would be great if there was a way to identify these types of issues in future. since it was not driving poorly, it was NOT driving.

bot66, 2022-05-02:
me too.I run carla simulator outside docker container, this problem still remains.

bot66, 2022-05-02:
same problem, how to fix it ?
[image]

ferryman0723, 2023-03-31:
> [ec7cd74](https://github.com/commaai/openpilot/commit/ec7cd74dc37caad9a281f70fa1bfcb12535efeb9) have the same problem, how to fix it ? [image]

@bot66 how did u fix it?I meet the same issue,thanks.