commaai/openpilot #24327: replay: fix CarParams write

sshane, 2022-04-25:
Currently it writes as Event.CarParams, but Python processes expect a CarParams struct

sshane, 2022-04-28:
@arudomanenko want to also try out this issue? If you try to start radard.py with replay you'll see what the issue is.

maykonpacheco, 2022-05-06:
@sshane I'm trying to solve this problem, could you explain how to run radard.py with replay.

sshane, 2022-05-06:
Start replay and then run radard.py, replay will stop publishing radarState packets whenever radard starts up, but you'll notice radard crashes on reading the CarParams param

maykonpacheco, 2022-05-11:
@sshane It seems like a very simple and silly question but this will help a lot of people who are trying to debug openpilot, could you explain and show how to read the CarParams parameter defined by replay?

sshane, 2022-05-12:
https://github.com/commaai/openpilot/blob/master/selfdrive/controls/radard.py#L187

Basically we need replay.cc to write the parameter in a way that radard will read it correctly.

closed 2022-05-22 by sshane