commaai/openpilot #24333: Chrysler ACC Fault when below 39mph

ns3284, 2022-04-26:
### Describe the bug

When cruise is set below 39mph, or ACC results in vehicle going below 39mph, and the driver touches the accelerator, the ACC system instantly faults.

Branch: master-ci

I will look into this issue, but wanted to open a ticket to reference. Appears there is a logic error when you mix in the accelerator press.

### What hardware does this issue affect?

comma three

### Which car does this affect?

2022 Chrysler Pacifica Hybrid 

### Provide a route where the issue occurs

N/A

### openpilot version

0.8.14b

adeebshihadeh, 2022-04-26:
We'll need a route where this occurs in order to look into it.

ns3284, 2022-04-26:
Sure. I'll have to pull one. I still haven't registered my device/account yet, so don't have a route. Once I do that, it will be extremely easy to repro.

ns3284, 2022-04-26:
@adeebshihadeh 4395faadf0a373f2|2022-04-26--13-42-38--0

ns3284, 2022-04-26:
Sure. I'll drive the vehicle in about 30mins and do that.

I'll go above min steer, cruise, hit gas, cancel. Then go below min steer, cruise hit gas, and it will fault. I believe it faults as soon as I press the accel, not when I let go.

ns3284, 2022-04-26:
Disregard, was above min speed, and still faulted when I pressed the accel. Will upload logs now.

ns3284, 2022-04-26:
4395faadf0a373f2|2022-04-26--15-41-55--0

ns3284, 2022-04-26:
> 

Ah, ok. I had Openpilot enabled.

ns3284, 2022-04-26:
4395faadf0a373f2|2022-04-26--15-47-07--0

Dashcam mode, set ACC above hit gas, set below, hit gas. My HUD on car shows ACC Driver Override when I press gas.

sshane, 2022-04-26:
Do you have a route with stock LKAS controlling steering? Not seeing any differences in the messages openpilot sends (LKAS_HUD and LKAS_COMMAND)

ns3284, 2022-04-26:
4395faadf0a373f2|2022-04-24--20-01-43--3

This one is where it is driving, but I do not think LKAS accelerated, just steered.

ns3284, 2022-04-26:
So I'm learning the tools now. I can see the jumbled LKAS messages in the qlog. What is the proper tool to view the LKAS commands?

ns3284, 2022-04-26:
I'm going to guess the problem is that LKAS probably gets temporarily disabled when you press the gas, so something like this might need to be added.

```
    gas_pressed = CS.out.gasPressed
    lkas_active = moving_fast and enabled and not gas_pressed
```

ns3284, 2022-04-26:
Right at the time I press the gas, while ACC is on, I also see an 
CS.out.events[1] change to name = 73, preEnable = 1
and
CS.out.events[2].name = 1, immediateDisable=1

After Gas Press:
[image]

Before Gas Press:
[image]

ns3284, 2022-04-26:
For event[1].name = 1, according to cereal, that looks to be "steerUnavailable".

```
  enum EventName @0xbaa8c5d505f727de {
    canError @0;
    steerUnavailable @1;
```

sshane, 2022-04-27:
That's an openpilot route, you want to turn `Enable openpilot` off in settings, then drive around on the stock ACC and LKAS systems and engage with LKAS steering for a bit.

ns3284, 2022-04-27:
So I think I found the problematic setting. "Disengage On Accelerator Pedal". I had it enabled, but that causes ACC Faults. Disabled it and no issues.

sshane, 2022-04-27:
Not really sure what's going on here, but try this branch: https://github.com/commaai/openpilot/tree/cleanup-chrysler-cc It should have been using latActive anyway.

Make sure you flip the Disengage On Accelerator Pedal toggle ***back*** to where it was causing faults, we want to resolve those.

ns3284, 2022-04-27:
> Accelerator

Could be that I didn't let it gather enough data. I'll switch to that branch, and put it back in the troublesome state, and fault ACC, the  wait for the Comma device to try and send new commands.

ns3284, 2022-04-27:
Still need to switch branches, will do that shortly, but gathered some more datapoints.

>4395faadf0a373f2|2022-04-27--08-56-20--2

This one I had it driving, accelerating by increasing speed via Cruise speed adjustments, and initiated a lane change. While it was lane changing, I tapped the gas, and the ACC faulted. Kept driving, but I think as far as Openpilot was concerned, I turned off Cruise Control.

>4395faadf0a373f2|2022-04-27--08-46-35--3

On this one, I did not fault the system, but played with the speed to accel/decel. No faults happened as I did not manually touch the gas pedal.

>4395faadf0a373f2|2022-04-27--08-33-59--9

This one is a short one, where I enabled, accelerated with controls (I believe), and touched the gas, and it faulted. I let it coast for a bit, but did not see any faults on the Comma device. Again, just behaving as if I disabled Cruise Control.

I'll switch to the other branch now.

ns3284, 2022-04-27:
>4395faadf0a373f2|2022-04-27--11-23-53--3

Another fault, this time I didn't touch the gas, but ACC was slowing down on its own. This is on the cleanup-chrysler-cc branch.

ns3284, 2022-04-27:
>4395faadf0a373f2|2022-04-27--12-03-04--5

One more, this time it didn't fault when above 40mph and I pressed the gas, but faulted below 39mph when I pressed the gas.

sshane, 2022-04-28:
`4395faadf0a373f2|2022-04-27--11-23-53--3`: It looks like ACC was enabled and then you pressed the gas to disengage, it was also keeping a constant speed until then.

One thing I'm noticing is ACC_2->ACC_STATUS_2 goes to 1 after your faults, when normally it's 3 when not engaged. This isn't definitive however as it's also 1 when you first start your car, so maybe it's an ACC available signal?

I pushed an update to that branch to use the counter from the car for the wheel buttons, it's possible the ACC system is sensitive to that and needs to be close to what it's expecting.

ns3284, 2022-04-28:
Sure. So by default, ACC is always off when I start the car. Usually I turn it on right away, and then set the speed after driving for a bit. I'll pull the latest from that branch and give it a run.

ns3284, 2022-04-28:
>4395faadf0a373f2|2022-04-28--08-56-33--5

I flipped the ACC on/off a bunch on this, actually hit the accel a few times and did not fault ACC. Eventually it did fault.

Resulting in this:
[image]
[image]
[image]

ns3284, 2022-04-28:
>4395faadf0a373f2|2022-04-28--09-07-39--2

This one faulted pretty quickly. I lowered set speed, was above that speed, and when ACC expected to coast down to set speed, I pressed gas pedal and it faulted.

ns3284, 2022-04-28:
As for just pressing buttons on the wheel...

>4395faadf0a373f2|2022-04-28--08-48-52--0
4395faadf0a373f2|2022-04-28--08-48-52--1

I pressed the buttons a lot on these routes, no faults though, and Cruise was never engaged.

ns3284, 2022-04-28:
So I think I pinpointed the in the logs when the fault hits, and this is what I see.

Just before at logMonoTime = 2793073360558
>...
    (address = 571, busTime = 0, dat = "\001P,", src = 128),
...
    ( address = 658,
      busTime = 0,
      dat = "\024\000\000\000@X",
      src = 128 ),

Followed by these sendcan messages. Possible the data is malformed, or redundant/conflicting with what the vehicle is expecting?

>( logMonoTime = 2793075600218,
  sendcan = [
    (address = 571, busTime = 0, dat = "\001`f", src = 0),
    ( address = 658,
      busTime = 0,
      dat = "\024\000\000\000P\x95",
      src = 0 ) ],
  valid = true )

sshane, 2022-04-28:
@ns3284 all of those drives are on the commit from two days ago, please make sure you're on the latest commit and try again. (check for update in device settings or git pull over ssh)

ns3284, 2022-04-28:
Hmm. I though I did that. Must have been user error. I did an ssh to the Comma device, git pull, systemctl reboot -i, and then went for the drive.

Possible it didn't actually reboot like I had thought.

```
comma@tici:/data/openpilot$ git status
On branch cleanup-chrysler-cc
Your branch is up to date with 'origin/cleanup-chrysler-cc'.

nothing to commit, working tree clean
comma@tici:/data/openpilot$ git pull
Already up to date.
comma@tici:/data/openpilot$ git log -1
commit 0c883a8a2f2c2085f8d80a63a1d8e667f05a10f6 (HEAD -> cleanup-chrysler-cc, origin/cleanup-chrysler-cc)
Author: Shane Smiskol <shane@smiskol.com>
Date:   Thu Apr 28 01:17:22 2022 -0700

    use wheel buttons counter and clean up
```

sshane, 2022-04-28:
You can just run `sudo reboot`, you need to enter a password for that command. Do you have a Discord? Perhaps that would be easier to debug this

closed 2022-05-07 by sshane