commaai/openpilot #24375: Car Unsupported

BadOlives, 2022-04-29:
### Describe the bug

Turn on the ignition, Comma 3 goes onroad and immediately greeted with Car Unsupported message for a few seconds, then switching to Calibration In-Progress screen.

I was able to pull the fingerprint from that drive:

`'carFingerprint': 'mock',
               'carFw': [{'address': 2016, 'ecu': 'engine', 'fwVersion': b'\xf1\x00\xa2\x10\t', 'subAddress': 0},
                         {'address': 2017, 'ecu': 'transmission', 'fwVersion': b'\xf1\x00\xa4\x10@', 'subAddress': 0},
                         {'address': 1862, 'ecu': 'eps', 'fwVersion': b'z\xc0\x04\x00', 'subAddress': 0},
                         {'address': 1927, 'ecu': 'fwdCamera', 'fwVersion': b'\x00\x00dd\x00\x00\x00\x00', 'subAddress': 0},
                         {'address': 1968, 'ecu': 'esp', 'fwVersion': b'\xf1\x00\xb2\x06\x04', 'subAddress': 0},
                         {'address': 2000, 'ecu': 'fwdRadar', 'fwVersion': b'\xf1\x00\x00\x00\x02', 'subAddress': 0},
                         {'address': 1988, 'ecu': 'fwdCamera', 'fwVersion': b'\xf1\x00\xac\x03\x00', 'subAddress': 0}],
               'carName': 'mock',`

### What hardware does this issue affect?

comma three

### Which car does this affect?

2018 Subaru Crosstrek Limited

### Provide a route where the issue occurs

N/A

### openpilot version

0.8.14 Master

sshane, 2022-04-29:
Do you have a route or dongle ID?

BadOlives, 2022-04-29:
> Do you have a route or dongle ID?

314d9ea40bdefcad|2022-04-29--15-47-24

sshane, 2022-04-30:
Can you try this branch out? https://github.com/commaai/openpilot/pull/24377

BadOlives, 2022-04-30:
> Can you try this branch out? #24377

Yessir, give me like 15 minutes.

BadOlives, 2022-04-30:
> Can you try this branch out? #24377

Took it for a short drive. Restarted the vehicle several times and had no issues with it detecting my vehicle. Good to go!

closed 2022-04-30 by BadOlives