commaai/openpilot #24408: modeld no longer draws path and lane lines on x86 (uiview.py)

zorrobyte, 2022-05-03:
### Describe the bug

After reverting the commit that caused the corrupted video feed in UI, I can't get the lane lines and path to render now:
[image]

* stock OP, latest master
* revert commit which broke video 
* run selfdrive/debug/uiview.py

modeld spams `selfdrive/modeld/modeld.cc: vipc_client_main no frame`, but this is expected according to https://github.com/commaai/openpilot/issues/24405

### OS Version

Ubuntu 20.04

### openpilot version or commit

05f606c8121c89990225ff873c2779fcb3790ead

zorrobyte, 2022-05-03:
Possibly due to https://github.com/commaai/openpilot/pull/23949/

pd0wm, 2022-05-04:
Modeld is expecting two camera feeds. You can try setting the `EnableWideCamera` param to use the wide camera for both feeds. Makee sure the wide camera intrinsics match your webcam.

zorrobyte, 2022-05-04:
> Modeld is expecting two camera feeds. You can try setting the `EnableWideCamera` param to use the wide camera for both feeds. Make sure the wide camera intrinsics match your webcam.

Thanks for that!

Also note that some tools are broken, like debug/uiview as they have not been updated with the wide camera feed. I'll go ahead and try this param for both use cases and report back. I kinda suck at C, but always a good time to learn!

zorrobyte, 2022-05-04:
Figured out that the TICI check exists elsewhere in the code and re-added it, also, the `use_extra_client` conditional was reversed.
https://github.com/commaai/openpilot/pull/24421

Still not getting the drawn path, but it's a step in the right direction

zorrobyte, 2022-05-04:
Fixed in https://github.com/commaai/openpilot/pull/24422

closed 2022-05-04 by zorrobyte