commaai/openpilot #24425: Master-ci: Laneless model crosses center double yellow lane lines

nickgeurnop, 2022-05-04:
### Describe the bug

As of writing, I am running the latest master-ci build on a 2019 Toyota Corolla Hatchback XSE with a comma three. During a leftward, sweeping turn on a 50mph two lane road, Openpilot crosses the center double yellow lanelines. Previous builds of master-ci 0.8.14 have not exhibited this behavior. 

Model: Laneless
Time: 16:25:50

### What hardware does this issue affect?

comma three

### Provide a route where the issue occurs

e9b1885ec511d838|2022-05-04--16-24-29--1

### openpilot version

master-ci 0.8.14

redacid95, 2022-05-04:
This is a duplicate report of #23853. Known issue that they are looking into with big model.

closed 2022-05-04 by adeebshihadeh

nickgeurnop, 2022-05-04:
My bad! Would it help for me to leave a comment on the original issue? Again, sorry about the oversight.