commaai/openpilot #24444: Toyota Highlander Hybrid 2021 "car not supported" after service yesterday (5/4/22)

cyberena, 2022-05-05:
### Describe the bug

Comma3 open pilot latest worked perfectly until service yesterday.  Guessing Toyotas latest updates made it hard to identify.

Any chance of getting fixed quickly in master?

### What hardware does this issue affect?

comma three

### Which car does this affect?

Toyota Highlander Hybrid Platinum 2021

### Provide a route where the issue occurs

all

### openpilot version

0.8.13

pd0wm, 2022-05-05:
What is your dongle id?

cyberena, 2022-05-05:
Hi, thanks for your reply.  I did reset the OS in an attempt to fix it, let me know if you need me to do anything

My dongle id is 382c11ef65f425c6

ErichMoraga, 2022-05-06:
Should just be the usual fingerprint issue.  We've been dealing with this quite a bit.

cyberena, 2022-05-06:
I see, so what should I do now?

ErichMoraga, 2022-05-06:
> I see, so what should I do now?
> […](#)

https://github.com/commaai/openpilot/wiki/Fingerprinting

If you need help with that, join us in the #toyota-lexus Discord channel where we handle this every day.

sshane, 2022-05-06:
It looks like your new esp fw was added in this PR by another user: https://github.com/commaai/openpilot/pull/24289

Until the release soon, try the branches master-ci or master

closed 2022-05-06 by sshane

cyberena, 2022-05-06:
Hi sshane, what does "esp fw" mean? 
What is the URL I should use to test the master-ci?

sshane, 2022-05-06:
That's a specific ECU in your car we use to fingerprint your vehicle. It just means it could have been reflashed when you got service done. You can uninstall openpilot and use the URL: https://installer.comma.ai/commaai/master-ci

cyberena, 2022-05-06:
OMG it worked, Thank you Shane!!  I was getting depressed thinking I lost my comma pilot for good lol

BTW, whats the coolest version I should try with all the latest features (even if its beta mode) :)

ErichMoraga, 2022-05-06:
See #custom-forks discussion in the Comma Discord.