commaai/openpilot #24456: 2019 Canadian prius fingerprinted as 2017, but appears to have different torque

HaraldSchafer, 2022-05-06:
### Describe the bug

These two TSSP priuses appear to have very different amount of torque available:
e6448e8ba54a1aab|2022-05-04--21-06-06

vs 2019 Canadian Prius
72c8dc9054796c61|2022-05-06--13-15-51

But are fingerprinted the same.

### What hardware does this issue affect?

comma three, comma two, EON Gold

### Which car does this affect?

Toyota 2017 prius

### Provide a route where the issue occurs

72c8dc9054796c61|2022-05-06--13-15-51

### openpilot version

master

sshane, 2022-05-06:
Looks like this car has a good angle sensor, per comment here: https://github.com/commaai/openpilot/blob/master/selfdrive/car/toyota/values.py#L1066

We check the EPS fw in interface for the RAV4, probably want to do something similar here

sshane, 2022-07-07:
Is this resolved in https://github.com/commaai/openpilot/commit/dbfe923ecc0e14a781a7f03646ccf885ab2fa47a?

HaraldSchafer, 2022-07-07:
No not yet, will likely need online torque value learner

adeebshihadeh, 2022-08-13:
Now that we know this is the case for a lot of cars and we're written the live learner, I think we can close this.

closed 2022-08-13 by adeebshihadeh