commaai/openpilot #24465: Panda Jungle doesn't work

zorrobyte, 2022-05-08:
### Describe the bug

Received my Panda Jungle and Black Panda today! Flashed the jungle, cloned the repo. Updated the black panda.

Pressing the ignition button lights the LED but doesn't do anything. `can_send.py` makes pretty lights turn on the jungle, the panda black heartbeats red, but nothing happens. I've tried on a few different platforms.

### OS Version

Ubuntu 20.04

### openpilot version or commit

ba89faa5f77c76023f49b85888e6d6eb7f2e1a11

### Additional info

Please let me know how I can help troubleshoot.

zorrobyte, 2022-05-08:
I downgraded to OP 0.8.10 and it works! It looks like some updates are going to be needed since CAN-FD was introduced.

adeebshihadeh, 2022-05-09:
We can't look into this issue. What do you expect the jungle to do? There's no script called `can_send.py` in this repo.

closed 2022-05-09 by adeebshihadeh

zorrobyte, 2022-05-09:
Sorry, can_replay.py
[image]

I would expect the jungle to at least "start" (trigger ignition) as on the storage page
[image]

@adeebshihadeh if you guys can't look into it, @geohot would you guys remove the jungle from sale? I'll need to return mine :(

adeebshihadeh, 2022-05-09:
It's not that we don't take issues for the jungle. I have several questions after reading this issue, the most troubling of which is: what is the issue? The jungle has many different use cases. What are you using it for? What debugging steps have been taken?

The issues you refer to often look more like support and help getting started, which isn't what our issue tracker is for. I'd recommend our Discussions page or Discord for that. I'd also recommend reading http://www.catb.org/~esr/faqs/smart-questions.html before creating future issues. The quality of the answer depends strongly on the quality of the question.

zorrobyte, 2022-05-09:
I appreciate your feedback, I'll strive to do better in the future