commaai/openpilot #24467: Webcam doesn't work

zorrobyte, 2022-05-08:
### Describe the bug

Since the C2/single cam removal (and https://github.com/commaai/openpilot/pull/24263), webcam no longer works following the comma zero guide/readme.

I was working on a PR, but I'm too stupid in C to submit something that isn't embarrassing. I put out some feelers for some help on Discord https://discordapp.com/channels/469524606043160576/524594418628558878/972533562555371570

I also didn't know if we just want to send the same frame to both the narrow and wide streams since we are forced to use both, or if we want to do something fancier, like send differing FoVs to the narrow/wide streams as the comma recommended webcam is 78 degrees FoV; which covers both model's inputs of 60 degrees in for wide cam and 30 degrees for narrow.

[image]

### OS Version

Ubuntu 20.04

### openpilot version or commit

191fbd4f67e7f878b518effd14411779b9435339

pd0wm, 2022-05-09:
By setting `EnableWideCamera` and sending the webcam frames over the wide angle steam you should be able to only use a single stream.

zorrobyte, 2022-05-09:
I am working on making this issue more actionable (http://www.catb.org/~esr/faqs/smart-questions.html) and will update the issue later today while I try to implement the suggestion.

closed 2022-05-15 by zorrobyte