commaai/openpilot #24487: CameraViewWidget assertion failed during replay

jeroenlammersma, 2022-05-10:
### Describe the bug

I am on the master branch, without any modifications.

When I start the demo replay using:
```selfdrive/ui/replay/replay --demo```
and then starting the ui using:
```selfdrive/ui/ui```

The following error occures:
```_ui: selfdrive/ui/qt/widgets/cameraview.cc:243: void CameraViewWidget::vipcConnected(VisionIpcClient *): Assertion `glGetError() == GL_NO_ERROR' failed.
Aborted (core dumped)```

### OS Version

Ubuntu 20,04

maykonpacheco, 2022-05-11:
@jeroenlammersma Did you follow these steps?
https://github.com/commaai/openpilot/blob/master/tools/README.md

[image]

jeroenlammersma, 2022-05-11:
Yes I did. And when I remove the assertion
```assert(glGetError() == GL_NO_ERROR);```
at lines 231, 235 and 260 from https://github.com/commaai/openpilot/blob/master/selfdrive/ui/qt/widgets/cameraview.cc
and then rebuild, it kinda works, but it seems it won't load the video stream correctly and this is what I get:
[image]

I also tried the debug ui
```python replay/ui.py```
and that one works without any issues.

pd0wm, 2022-05-11:
Can you printing the exact openGL using `glGetError`? Are other opengl applications working correctly on your system?

khoi, 2022-05-11:
https://github.com/commaai/openpilot/pull/24317/files seems to break the build for macOS Silicon as well. I'm also getting `glGetError`

jeroenlammersma, 2022-05-11:
@pd0wm

When I run
```glxinfo | grep 'version'```

I get:
```
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
    Max core profile version: 4.1
    Max compat profile version: 4.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 2.0
OpenGL core profile version string: 4.1 (Core Profile) Mesa 21.2.6
OpenGL core profile shading language version string: 4.10
OpenGL version string: 4.1 (Compatibility Profile) Mesa 21.2.6
OpenGL shading language version string: 4.10
OpenGL ES profile version string: OpenGL ES 2.0 Mesa 21.2.6
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 1.0.16```

I have not tested other opengl applications, because I am not sure which applications use it. Do you know a one I can try?

pd0wm, 2022-05-11:
You can try something like glxgears. What hardware are you using?

The real way to debug this is to add glGetError and see what exact error you're hitting.

jeroenlammersma, 2022-05-11:
I added that snippet to the code. When executing I receive like a dozen of the following error:
```gl error: 0```

The console:
[image]

I had some issues with the rig running dedicated Ubuntu, but I will have it running tomorrow and will post the results here then.

Thanks anyway already! I hope it is just a VM issue.

jeroenlammersma, 2022-05-11:
But when thinking about it a bit more ```GL_NO_ERROR``` is ```0``` right?

because in gl3.h GL_NO_ERROR is defined as 0
```#define GL_NO_ERROR                       0```

So the assert should actually not fail right?
I am confused now :)

Ok, I will try on the dedicated Ubuntu machine tomorrow and see how it goes.

pd0wm, 2022-05-12:
discord.comma.ai is probably faster if you need troubleshooting help (see the #dev-openpilot channel). Feel free to open a new issue if there is an actual bug.

closed 2022-05-12 by pd0wm

jeroenlammersma, 2022-05-12:
@pd0wm 
I did some experimenting and I found out you just need adequate hardware for it to run properly. The VM on my laptop just didn't had enough. Thanks for help anyway and will go to discord next time :)

pd0wm, 2022-05-12:
About the `glGetError()`. I just noticed calling that function clears the error. You need to store it in a variable if you want to check and print it.