commaai/openpilot #24500: build error on pc

chenti2x, 2022-05-12:
### Describe the bug

run scons -u -j$(nproc),the error is this:

In file included from panda/board/stm32fx/inc/core_cm4.h:34:0,
                 from panda/board/stm32fx/inc/stm32f413xx.h:197,
                 from panda/board/stm32fx/inc/stm32f4xx.h:189,
                 from panda/board/stm32fx/stm32fx_config.h:2,
                 from panda/board/config.h:34,
                 from panda/board/bootstub.c:7:
/usr/lib/gcc/arm-none-eabi/6.3.1/include/stdint.h:9:26: fatal error: stdint.h: No such file or directory
 # include_next <stdint.h>
                          ^
compilation terminated.
scons: *** [panda/board/bootstub-panda.o] Error 1
In file included from panda/board/stm32h7/inc/core_cm7.h:34:0,
                 from panda/board/stm32h7/inc/stm32h725xx.h:226,
                 from panda/board/stm32h7/inc/stm32h7xx.h:160,
                 from panda/board/stm32h7/stm32h7_config.h:1,
                 from panda/board/config.h:32,
                 from panda/board/bootstub.c:7:
/usr/lib/gcc/arm-none-eabi/6.3.1/include/stdint.h:9:26: fatal error: stdint.h: No such file or directory
 # include_next <stdint.h>
                          ^
compilation terminated.
scons: *** [panda/board/bootstub-panda_h7.o] Error 1
Compiling /home/ada/newdisk/ctt/openpilot/opendbc/can/parser_pyx.pyx because it changed.
[1/1] Cythonizing /home/ada/newdisk/ctt/openpilot/opendbc/can/parser_pyx.pyx
Compiling /home/ada/newdisk/ctt/openpilot/opendbc/can/packer_pyx.pyx because it changed.
[1/1] Cythonizing /home/ada/newdisk/ctt/openpilot/opendbc/can/packer_pyx.pyx
scons: building terminated because of errors.

### OS Version

ubuntu18.04

### openpilot version or commit

master

pd0wm, 2022-05-12:
Ubuntu 18.04 is not supported. Try ubuntu 20.04 or 22.04

closed 2022-05-12 by pd0wm

chenti2x, 2022-05-13:
hello，can I compile and run on ubuntu18.04 with ubuntu20.04 environment in docker?Or do I have to install an ubuntu20.04 environment on the computer?

lidongyv, 2022-06-13:
@chenti2x 
you don't need correctly set up the env on pc if you only want to run the openpiot in docker