commaai/openpilot #24536: error  when run the model supercombo.onnx in onnx_runner.py

fatsheep2020, 2022-05-14:
### Describe the bug

run carla and OP container in PC , there is error report: [AssertionError]
[image]

How to reproduce:
sudo INSTALL=1 ./start_carla.sh
enter openpilot container, then run launch_openpilot.sh
there will be a error log as above pic show.

### OS Version

Ubuntu 20.04

### openpilot version or commit

v0.8.14

### Additional info

this error will lead to 'movelV2' not alive, and after some study, this error happen when run the model supercombo.onnx in onnx_runner.py, at same time, check the file, it exist:
[image]

maykonpacheco, 2022-05-17:
I have the same problem.

maykonpacheco, 2022-05-17:
[image]

maykonpacheco, 2022-05-21:
#24576

fatsheep2020, 2022-05-22:
not exactly same issue, what I meet is there is model (onnx) read error which happened in onnx_runner.py  @maykonpacheco

fatsheep2020, 2022-05-22:
and the HMI have issue #24232

maykonpacheco, 2022-05-22:
In addition to the enconded error, mine also has this error in the model @fatsheep2020

gijskoning, 2022-05-29:
Hey, these errors have different causes. The encoderd one is legit and will be fixed with https://github.com/commaai/openpilot/pull/24676. 
@fatsheep2020 I am not sure how to fix your issue. Did you do all the correct steps to install openpilot that are listed in the readme?

maykonpacheco, 2022-05-31:
I have the same problem.
[image]

HaraldSchafer, 2022-06-03:
I could not reproduce this @maykonpacheco You still have this on latest master?

closed 2022-06-12 by adeebshihadeh