commaai/openpilot #24559: Radarless Honda: enable longitudinal

sshane, 2022-05-17:
On these cars (2022 Civic), the camera seems to send the ACC messages on address `0x1C8`.

Need to find the bits for AEB since the current `ACC_CONTROL` (`0x1DD`) had those and the DBC definition for it doesn't look correct at all.

kevinharbin, 2022-05-23:
Might be here 
https://cabana.comma.ai/?route=f44aa96ace22f34a%7C2022-05-23--16-17-30&seekTime=1&url=https%3A%2F%2Fchffrprivate.azureedge.net%2Fchffrprivate3%2Fv2%2Ff44aa96ace22f34a%2Fcf8616c5c88a4736cea2b14aa073b41b_2022-05-23--16-17-30
[image]

sshane, 2023-03-28:
Done in https://github.com/commaai/openpilot/pull/25364

closed 2023-03-28 by sshane