commaai/openpilot #24570: Remove thumbnails

geohot, 2022-05-17:
These things are in camerad and terrible. They run in the frame loop and waste CPU. My proposal is this:

Every 15 frames, we have an IFrame in the qcamera stream. These are about 5kb and qcamera quality. We could add the EncodeData packets for these to the logs instead (maybe every 180 frames aka 9 seconds, do we really need 5?), then change UploadQlogJpegs to decode them in ffmpeg.

gregjhogan, 2022-05-18:
Is CPU really a concern if we knock it down to one per segment?  We discussed this previously and I believe we concluded we have way more than we need and they are way bigger then they need to be.

geohot, 2022-05-19:
It's not just CPU, it's sporadic CPU. I believe most of the time is spent in the downscale actually, so smaller wouldn't even help.

sshane, 2024-07-20:
Now that we don't upload qcams by default on cell, we can't use the I-frames to show thumbnails before qcams upload, which we still want.

1) generate thumbnails in encoderd or loggerd (so loop of camerad isn't variable depending on if it's making a thumbnail or not)
2) reduce resolution and frequency. 1/8th and 2 per segment? currently it's 1/4th and 12 per segment

closed 2025-02-10 by adeebshihadeh