commaai/openpilot #24576: CARLA simulator broken after 'Add desired curvature to controlsState' commit

jeroenlammersma, 2022-05-18:
### Describe the bug

From commit #6518ecc 'Add desired curvature to controlsState' the CARLA simulator is broken.
When launched, modeld and controlsd services stop working and openpilot is unusable inside the simulator.

In the additional info, you can find the full log.
On the last line, the services that are not responding/working are modeld and controlsd (surrounded with stars: **<service>**).

### OS Version

Ubuntu 20.04

### Additional info

```
./modeld: 3: Bad substitution
Waiting for CAN messages...
clocksd logcatd modeld navd proclogd ui soundd locationd calibrationd controlsd deleter logmessaged paramsd plannerd radard thermald statsd
VIN 00000000000000000
platform[0] CL_PLATFORM_NAME: NVIDIA CUDA
vendor: NVIDIA Corporation
platform version: OpenCL 3.0 CUDA 11.6.134
profile: FULL_PROFILE
extensions: cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_3d_image_writes cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_copy_opts cl_nv_create_buffer cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_device_uuid cl_khr_pci_bus_info cl_khr_external_semaphore cl_khr_external_memory cl_khr_external_semaphore_opaque_fd cl_khr_external_memory_opaque_fd
name :NVIDIA GeForce RTX 3070
device version :OpenCL 3.0 CUDA
max work group size :1024
type = 4 = CL_DEVICE_TYPE_GPU
selfdrive/modeld/modeld.cc: models loaded, modeld starting
selfdrive/modeld/modeld.cc: connected main cam with buffer size: 3493536 (1928 x 1208)
selfdrive/modeld/modeld.cc: connected extra cam with buffer size: 3493536 (1928 x 1208)
selfdrive/modeld/modeld.cc: skipping model eval. Dropped 13 frames
--- Logging error ---
Traceback (most recent call last):
  File "/home/jeroen/.pyenv/versions/3.8.10/lib/python3.8/logging/__init__.py", line 1085, in emit
    msg = self.format(record)
  File "/home/jeroen/.pyenv/versions/3.8.10/lib/python3.8/logging/__init__.py", line 929, in format
    return fmt.format(record)
  File "/home/jeroen/openpilot/common/logging_extra.py", line 100, in format
    v = json.loads(record)
  File "/home/jeroen/.pyenv/versions/3.8.10/lib/python3.8/json/__init__.py", line 357, in loads
    return _default_decoder.decode(s)
  File "/home/jeroen/.pyenv/versions/3.8.10/lib/python3.8/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/home/jeroen/.pyenv/versions/3.8.10/lib/python3.8/json/decoder.py", line 353, in raw_decode
    obj, end = self.scan_once(s, idx)
json.decoder.JSONDecodeError: Expecting property name enclosed in double quotes: line 1 column 24 (char 23)
Call stack:
  File "/home/jeroen/openpilot/selfdrive/manager/manager.py", line 205, in <module>
    main()
  File "/home/jeroen/openpilot/selfdrive/manager/manager.py", line 182, in main
    manager_thread()
  File "/home/jeroen/openpilot/selfdrive/manager/manager.py", line 134, in manager_thread
    ensure_running(managed_processes.values(), False, params=params, CP=sm['carParams'], not_run=ignore)
  File "/home/jeroen/openpilot/selfdrive/manager/process.py", line 313, in ensure_running
    p.start()
  File "/home/jeroen/openpilot/selfdrive/manager/process.py", line 245, in start
    self.proc.start()
  File "/home/jeroen/.pyenv/versions/3.8.10/lib/python3.8/multiprocessing/process.py", line 121, in start
    self._popen = self._Popen(self)
  File "/home/jeroen/.pyenv/versions/3.8.10/lib/python3.8/multiprocessing/context.py", line 224, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "/home/jeroen/.pyenv/versions/3.8.10/lib/python3.8/multiprocessing/context.py", line 277, in _Popen
    return Popen(process_obj)
  File "/home/jeroen/.pyenv/versions/3.8.10/lib/python3.8/multiprocessing/popen_fork.py", line 19, in __init__
    self._launch(process_obj)
  File "/home/jeroen/.pyenv/versions/3.8.10/lib/python3.8/multiprocessing/popen_fork.py", line 75, in _launch
    code = process_obj._bootstrap(parent_sentinel=child_r)
  File "/home/jeroen/.pyenv/versions/3.8.10/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
[... 192 lines cut ...]
  File "/home/jeroen/openpilot/selfdrive/controls/controlsd.py", line 816, in controlsd_thread
    self.step()
  File "/home/jeroen/openpilot/selfdrive/controls/controlsd.py", line 808, in step
    self.publish_logs(CS, start_time, CC, lac_log)
  File "/home/jeroen/openpilot/selfdrive/controls/controlsd.py", line 726, in publish_logs
    controlsState.desiredCurvature = self.desired_curvature
  File "capnp/lib/capnp.pyx", line 1377, in capnp.lib.capnp._DynamicStructBuilder.__setattr__
  File "capnp/lib/capnp.pyx", line 1375, in capnp.lib.capnp._DynamicStructBuilder.__setattr__
  File "capnp/lib/capnp.pyx", line 1368, in capnp.lib.capnp._DynamicStructBuilder._set
  File "capnp/lib/capnp.pyx", line 789, in capnp.lib.capnp._setDynamicField
AttributeError: capnp/schema.c++:493: failed: struct has no such member; name = desiredCurvature
stack: 7f46653a312f 7f4665282e5e 7f4665285e85 7f466523a7ac 7f4666631da2 7f46665bbcc0 7f46665b619a 7f46665b74cc 7f46665bb4f3 7f46665b619a 7f46665b74cc 7f46665bb4f3 7f46665b619a 7f46665b74cc 7f46665bb4f3 7f46665b619a 7f46665b74cc 7f46665baa94 7f46665b619a 7f46665e6312 7f46665b86de 7f46665b619a 7f46665b74cc 7f46665bb4f3 7f46666caa4a 7f46665e4273 7f46665e7c49 7f46665b74cc 7f46665b8c53 7f46665b619a 7f46665b74cc
Onnx available providers:  ['TensorrtExecutionProvider', 'CUDAExecutionProvider', 'CPUExecutionProvider']
Onnx selected provider:  ['CUDAExecutionProvider']
Traceback (most recent call last):
  File "/home/jeroen/openpilot/selfdrive/modeld/runners/onnx_runner.py", line 63, in <module>
    ort_session = ort.InferenceSession(sys.argv[1], options, providers=[provider])
  File "/home/jeroen/.local/share/virtualenvs/openpilot-JMDOVsY3/lib/python3.8/site-packages/onnxruntime/capi/onnxruntime_inference_collection.py", line 335, in __init__
    self._create_inference_session(providers, provider_options, disabled_optimizers)
  File "/home/jeroen/.local/share/virtualenvs/openpilot-JMDOVsY3/lib/python3.8/site-packages/onnxruntime/capi/onnxruntime_inference_collection.py", line 370, in _create_inference_session
    sess = C.InferenceSession(session_options, self._model_path, True, self._read_config_from_model)
onnxruntime.capi.onnxruntime_pybind11_state.InvalidProtobuf: [ONNXRuntimeError] : 7 : INVALID_PROTOBUF : Load model from ../../models/supercombo.onnx failed:Protobuf parsing failed.
_modeld: selfdrive/modeld/runners/onnxmodel.cc:62: void ONNXModel::pwrite(float *, int): Assertion `err >= 0' failed.
clocksd logcatd **modeld** navd proclogd ui soundd locationd calibrationd **controlsd** deleter logmessaged paramsd plannerd radard thermald statsd
```

sshane, 2022-05-18:
Have you updated your submodules and rebuilt?

maykonpacheco, 2022-05-20:
I have same problem.

maxy0524, 2022-05-25:
> @sshane Yes, I have updated my submodules and rebuilt openpilot. Luckily, **modeld** and **controlsd** do not cause problems anymore. However, **encoderd** is currently the service that is a cause for an issue.
> 
> I tried running the simulator using the docker image `./start_openpilot_docker.sh` and by running the local scripts `./launch_openpilot.sh` and `./bridge.py`, but they have the same result of having **encoderd** not working properly.
> 
> See the screenshots below: [image] [image]
> 
> At the moment of typing, it was a build of master branch from commit #fd63de7, because this was the most recent commit as of testing (so I cannot be 100% sure the most recent build as of now fixes the issue, but looking at the commits, the issue is most probably not fixed yet).

Have you solved your problem? I have the same problem.

maxy0524, 2022-05-25:
> I have same problem.

Have you solved your problem? I have the same problem

educandela1, 2022-05-26:
I have the same problem too!

fatsheep2020, 2022-05-28:
same problem too

maykonpacheco, 2022-05-29:
I believe the error is here.

[image]

maykonpacheco, 2022-05-29:
[image]

adeebshihadeh, 2022-06-02:
Should be fixed by 44bffc1b3b2c8a3f661d8ab28ee8765a1bc2219c

closed 2022-06-02 by adeebshihadeh