commaai/openpilot #24590: Add back webcam support in new daemon

adeebshihadeh, 2022-05-18:
The replay and webcam cameras were deleted in #24554 in an effort to re-focus camerad on being as clean, stable, and performant as possible on the comm three. The replay camera had already been superseded by the new replay tool, but the webcam . Now that visionipc has a nice python interface, I think it makes sense to re-write the webcam version of camerad as a small python daemon, which should also make it easier to hack on to support different configurations. This already works well enough for the CARLA simulator bridge.

youdaorobot, 2023-05-26:
I also want to know in current period, is there still a way to use PC+webcam running OP?

youdaorobot, 2023-06-07:
Or which branch can be used now to try webcam?

edgarwideman, 2023-07-22:
Also waiting

0312birdzhang, 2023-10-10:
Do we have some demos with new visionipc interface? I tried the [camerad.py](https://github.com/commaai/openpilot/blob/master/tools/sim/lib/camerad.py), seems not works well when test driver monitor.

bauer-san, 2023-10-14:
> Do we have some demos with new visionipc interface? I tried the [camerad.py](https://github.com/commaai/openpilot/blob/master/tools/sim/lib/camerad.py), seems not works well when test driver monitor.

@0312birdzhang Can you share your attempt to use camerad.py?
I wonder if using visionipc still the right choice?  I noticed [this Bounty](https://github.com/commaai/openpilot/issues/1038) - now closed and undone but maybe still a goal, @geohot ?

0312birdzhang, 2023-10-14:
> @0312birdzhang Can you share your attempt to use camerad.py?

I'm porting openpilot to rk3588s dev board, the cameras are big problem.

jimbrend, 2023-12-20:
on what model was it most stable?

Is there any idea what will be included with qualcomm devices?

closed 2024-01-11 by adeebshihadeh