commaai/openpilot #24623: track mypy coverage

pd0wm, 2022-05-23:
> It would also be nice if there was some way to verify that our coverage isn't going down over time. It seems like subtle changes can drastically affect the mypy coverage silently.
https://github.com/commaai/openpilot/issues/24277

JacobO1994, 2022-06-02:
Any thoughts on how the team prefers this to be done? I wouldn't mind trying to work this out.

devhsal, 2022-07-25:
Not sure if this is what you are looking for but [coveralls.io](https://coveralls.io/) is a service that provides real-time code-coverage statistics on all your repositories, total percentages covered and lines covered.

It's free for open-source projects. 

Here is an example from 2016 of the team at Zulip using the service to [achieve 100% mypy code coverage](https://blog.zulip.com/2016/10/13/static-types-in-python-oh-mypy/).

adeebshihadeh, 2023-08-22:
We just need stricter enforcement (#29458), this is a waste of time.

closed 2023-08-22 by adeebshihadeh