commaai/openpilot #24638: UI: better lane change alert

adeebshihadeh, 2022-05-24:


jozef-simo, 2022-05-24:
Initial UI Proposal ([Figma](https://www.figma.com/file/kPvxp4UcTTU9RLLOY7R2b9/openpilot---REDESIGN?node-id=303%3A3))

1 – STATE: "Changing Lanes" :
- VARIANT 1 - icon positioned next to the speed dial
- VARIANT 2 - since Variant 1 layout breaks UI in split-screen Navigation view, the icon in Variant 2 is centered and positioned by screen edges.

Preview:
VARIANT 1
[image]
VARIANT 2
[image]

2 – STATE: "Car Detected in Blindspot" *work in progress 🚧

ebranga, 2022-05-24:
I think "Blind VARIANT 2" is the best for blind spot.

BadOlives, 2022-05-31:
Very nice proposal. I too agree with Variant 2 for "Car Detected in Blind Spot".

JoneKone, 2022-06-01:
Both to variant 2 Although i feel when it's blocked the blinking ball could become larger and the next lane should be all red.

ebranga, 2022-06-01:
@Achilles308. Changing Lanes variant 2 is when the screen is split into Navigation and Camera Views.

jozef-simo, 2022-06-03:
Updated UI

State 1: Start Lane Change Once Safe

| Without Text Alert | With Text Alert |
| ------------- | ------------- |
| [image] | [image] |

State 2: Car Detected in Blindspot
| Without Text Alert | With Text Alert |
| ------------- | ------------- |
| [image] | [image] |
----

**QUESTION: With/Without text Alert?**
The current UI presents a default openpilot alert. It covers a substantial part of the screen and hides the driving path. however, the benefit of text alert is that it better highlights two points: 
- you (the driver) have to nudge to initiate the lane change
- it's the driver's responsibility to check if it's safe

Including the text alert - does it add contextual value or not? Should we prioritize our user groups differently? Experienced OP driver (might not need a text alert), novice OP driver (text alert might add additional context) in this regard?

Achilles308, 2022-06-03:
One thing I just realized... aren't there 5 states of lane change?

A) Pre lane change, the turn signal is activated
B) Lane change started and in progress, the wheel was nudged the correct direction indicating the lane is clear and OP starts to initiate the controls to do the lane change
C) Lange change complete, kind of a non step if you just hide the lane change UI elements upon completion

The other states are:
D) Lane change is blocked, the blind spot monitoring or enhanced DM indicated the lane is not clear
E) Invalid lane to change into, no valid lane is seen so no lane change is possible

Some ideas on how to implement something to indicate if you're between A or B as listed above:

When in pre lane change (A), maybe the arrow could be a light gray color and when the lane change is started and in progress (B) it then turns white. When the lane change is blocked (D), the arrow could remain light gray in the red circle of course

I prefer with text but I think I'd propose an additional option of how to implement the text. I think a VERY thin box at the bottom would suffice. Maybe the text height plus padding above and below in the vertical like 5 pixels or something absurdly small. Keep it very sleek for daily use. If a user isn't sure what's going on, the text will offer some clues and clarity. I think no text would work for most frequent users, but sometimes someone not as familiar will drive the car and they should have the opportunity to learn the system by having text feedback. I don't think it's toggle worthy either.

samcooler, 2022-06-03:
a small note: relying on red/green contrast or low-opacity red overlay for an important safety message is not [colorblind compatible design](https://www.google.com/search?q=red+green+colorblind+design). There should be a shape-based change (or additional symbol) in addition to a color change. Suggestion: the arrow shape could change to an X, or the icon circle shape could change to a larger X. The text overlay is also a decent improvement on the color-change-alone design.

jozef-simo, 2022-06-04:
UPDATE:
1. State 1/2 - Reduced padding on Alert text. Making it less obstructive for the daily driver.
2. State 2 (Car Detected in Blindspot) - Added car icon as an additional visual clue.

PREVIEW:

| State 1 | State 2 |
| ------------- | ------------- |
| [image] | [image] |

PREVIEW: State 2 (Car Detected in Blindspot) – Once the adjacent line clears out, the indicator turns green.
[image]

jozef-simo, 2022-06-04:
Following up @Achilles308 
> For Car Detected in Blindspot variant 2, instead of a full path turning red, can it be made so about a car length turns red instead? Would give better spacial indication of roughly where the car blocking the lane change is. In instances where the user is trying to change into a not valid lane, the full path red might be a useful distinction.

Would be nice to see it from behind the wheel (in action).

| State 2 - full path | State 2 - partial path |
| ------------- | ------------- |
| [image] | [image] |

BadOlives, 2022-06-05:
> Following up @Achilles308
>
>
> Would be nice to see it from behind the wheel (in action).
>
> State 2 - full path	State 2 - partial path
> [image]	[image]

That looks really good! I'm torn between the full path and partial path as both of them look amazing. Also, great work on the alerts at the bottom. I'd love to see this implemented for all alerts besides lane changes. It looks sleek and doesn't block half the screen.

Achilles308, 2022-06-07:
@jozef-simo I'm loving the improvements!
 
I think this alert text is a good size, but I'm curious what **bold** would look like with an even narrower padding? Like maybe half that padding even? In my mind, this text is going to be read like 1 in 100 times, so minimal is better. If this is for ALL alerts then maybe it's perfect or could even go bigger if the goal is constancy.

I really like the partial path being red, however, I thought of something else :D   What if it rendered the front bumper, wheel,  and partial hood of a generic looking car in that semi translucent red?

Like the front of something like this...
[image]

Achilles308, 2022-07-18:
What a terrible way to explain the concept... haha... I finally had time to mock up something better.... here's my concept as listed above about a transparent car front end shape in the lane that's being blocked for lane change.

[image]

haraschax, 2023-06-29:
Different design that works better when map view is open.

This would be pre-lanechange state:
[image]

And changing lane state:
[image]

@deanlee You want to try implement this?

closed 2024-01-05 by adeebshihadeh