commaai/openpilot #24645: Why doesn't the interface of openpilot not updated in the CARLA simulator?

maxy0524, 2022-05-24:
### Describe the bug

Openpilot always shows the initial interface, like 0 MPH, and the entire interface doesn't change at all. Why?

--------------# Terminal 1-----------------
((openpilot) ) ubuntu@VM-0-16-ubuntu:~/openpilot/tools/sim$ sudo ./start_carla.sh 

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

0.9.12: Pulling from carlasim/carla
Digest: sha256:287964782500baf48ee4fac84e55255306666f786f27673c8960ce042f92d892
Status: Image is up to date for carlasim/carla:0.9.12
docker.io/carlasim/carla:0.9.12
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found

--------------# Terminal 2  ./launch_openpilot.sh-----------------
selfdrive/ui/qt/onroad.cc: slow frame rate: 11.72 fps
selfdrive/ui/qt/onroad.cc: slow frame rate: 11.68 fps
selfdrive/ui/qt/onroad.cc: slow frame rate: 11.68 fps
selfdrive/ui/qt/onroad.cc: slow frame rate: 11.69 fps
selfdrive/ui/qt/onroad.cc: slow frame rate: 11.72 fps
selfdrive/ui/qt/onroad.cc: slow frame rate: 11.72 fps
clocksd logcatd encoderd modeld navd proclogd ui soundd locationd calibrationd controlsd deleter logmessaged paramsd plannerd radard thermald statsd
selfdrive/ui/qt/onroad.cc: slow frame rate: 11.72 fps
selfdrive/ui/qt/onroad.cc: slow frame rate: 11.77 fps
{"event": "commIssue", "error": true, "invalid": ["radarState"], "not_alive": ["modelV2", "longitudinalPlan", "lateralPlan", "liveLocationKalman", "liveParameters", "driverCameraState"], "not_freq_ok": ["modelV2", "longitudinalPlan", "lateralPlan", "liveLocationKalman", "liveParameters", "driverCameraState"], "can_error": true}
{"event": "commIssue", "error": true, "invalid": ["radarState"], "not_alive": ["modelV2", "longitudinalPlan", "lateralPlan", "liveLocationKalman", "liveParameters", "driverCameraState"], "not_freq_ok": ["modelV2", "longitudinalPlan", "lateralPlan", "liveLocationKalman", "liveParameters", "driverCameraState"], "can_error": false}
selfdrive/ui/qt/onroad.cc: slow frame rate: 11.77 fps
selfdrive/ui/qt/onroad.cc: slow frame rate: 11.72 fps
selfdrive/ui/qt/onroad.cc: slow frame rate: 11.75 fps
selfdrive/ui/qt/onroad.cc: slow frame rate: 11.78 fps
clocksd logcatd encoderd modeld navd proclogd ui soundd locationd calibrationd controlsd deleter logmessaged paramsd plannerd radard thermald statsd
selfdrive/ui/qt/onroad.cc: slow frame rate: 11.78 fps
selfdrive/ui/qt/onroad.cc: slow frame rate: 11.76 fps
selfdrive/ui/qt/onroad.cc: slow frame rate: 11.74 fps
selfdrive/ui/qt/onroad.cc: slow frame rate: 11.73 fps
selfdrive/ui/qt/onroad.cc: slow frame rate: 11.73 fps

--------------# Terminal 3  ./bridge.py-----------------
python3 ./bridge.py lagging by 274985.67 ms
python3 ./bridge.py lagging by 274975.75 ms
frame:  engaged: False ; throttle:  0.0 ; steer(c/deg):  -0.0 0.0 ; brake:  0.0
python3 ./bridge.py lagging by 275011.02 ms
python3 ./bridge.py lagging by 275001.26 ms
python3 ./bridge.py lagging by 274992.32 ms
python3 ./bridge.py lagging by 274982.67 ms
python3 ./bridge.py lagging by 274972.82 ms
python3 ./bridge.py lagging by 275003.99 ms
python3 ./bridge.py lagging by 274994.51 ms
python3 ./bridge.py lagging by 274984.72 ms

### OS Version

Ubuntu 20.04

maxy0524, 2022-05-24:
My cloud Server machine is "Ubuntu Server 20.04", and the graphics is Tesla T4. I connect to this computer through "Microsoft Remote Desktop Beta".

gijskoning, 2022-05-29:
From the lagging messages it seems that bridge.py doesn't work correctly. There is probably an error shown at the start of the process. Could you check what that is?

adeebshihadeh, 2022-06-02:
@gijskoning is right, this doesn't sound like an openpilot issue. Check out Discord or GitHub Discussions for help getting setup with CARLA.

closed 2022-06-02 by adeebshihadeh

lidongyv, 2022-06-09:
same problem in ubuntu 20 but works correctly in ubuntu 18, there must be some bug in the sim docker env,

limanman1990, 2023-02-21:
I meet the same problem. is there any body solved this?

ferryman0723, 2023-03-12:
> I meet the same problem. is there any body solved this?

I met it too,did you solve it ?

limanman1990, 2023-03-12:
no.

---- Replied Message ----
| From | ***@***.***> |
| Date | 03/12/2023 21:13 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [commaai/openpilot] Why doesn't the interface of openpilot not updated in the CARLA simulator? (Issue #24645) |

I meet the same problem. is there any body solved this?

I met it too,did you solve it ?

ferryman0723, 2023-03-19:
> no.
> […](#)

I use the both official docker and dockerfile to build docker,it met the same controls lagging problem